KimJorgensen/KungFuFlash #83: Firmware update with FAT16 SD card fails

cosmo6128, 2021-02-03:
Hi

As per issue subject, I've been using a FAT16 Sandisk 4GB (class 4) MicroSD card with my KFF that I got via an ebay seller which works fine to load/run PRGs and CRT files, but trying the firmware update process wouldn't recognise the UPD firmware files after I downloaded them and copied to the card using my Win7 x64 PC. The KFF launcher will only display the error message "File is not supported, or invalid".

I've tried the firmware update with a UPD file within a subfolder, from the root folder (after copying games etc to various folders on the card first), and finally with a blank FAT16 SD card, but none worked. I thought it's an issue with my PAL C64c (even though PRGs and CRTs are fine), so I'd tried it the firmware update on my PAL C64 breadbin (rev 250407) and also my PAL SX64. No luck there. Finally, I tried to update the KFF firmware with a Transcend 2GB MicroSD card with FAT16 format, which also didn't work. Finally, reformatting the 2GB card to FAT32, and then copied the UPD firmware files to the root of the card, and the launcher was then able to program the new firmware to the KFF.

 At the moment there's no mention of having to use only FAT32 format with MicroSD cards, so I interpreted "FAT format" as meaning FAT16 format, which seems ok for small SD cards (2GB/4GB). Could the text of the upgrade screen be updated to mention FAT32 format is required? Or could the KFF firmware updater be revised to support loading the UPD file from both FAT16 and FAT32 formats?

KimJorgensen, 2021-02-03:
Thank you for your report.

The reason that Kung Fu Flash does not say FAT16 or FAT32 is that it should work with both. I just tried with an old 2GB card and reformatted that to FAT16 and had no problems with the firmware update, so I'm not sure why this is not working for you. 
So I have a couple of questions:
Was the filename altered in any way? Did you safely eject the SD card after writing the UPD file? Can you read the UPD file back on the PC and validate that it is identical to the downloaded file? Was there problems with other files that you had written to the SD card (PRG/CRT)?

cosmo6128, 2021-02-04:
Hi Kim

thanks for checking - to respond to your questions:
1) no - the UPD firmware filename wasn't renamed (the extension wasn't changed either)
2) yes, normally I do use the safely remove and eject option in the Windows system tray.
3) I've checked this by using Total Commander to compare the UPD files on the MicroSD card with the downloaded UPD files, and no differences were found between them
4) no, as described earlier I haven't found any errors with lauching PRG files or CRT files (ie normal/classic ones or Easyflash CRTs).

Could there be an interaction with the Allocation Unit Size when I format the SD card to FAT16? Is there a specific size that you used in your FAT16 tests that I can check as well? 

Also, I've attached a picture of the SX64's screen that's displaying the error message of my KFF loaded with my 4GB FAT16 MicroSD card.

[image]

KimJorgensen, 2021-02-04:
Thank you for info.
I did try to change the allocation unit size from 32 kb to 64 kb and then I could also reproduce the error.
There was a bug in the SD card handling code that was triggered by the file size of the upd file which is exactly 64 kb. I'll get the bug fixed.

KimJorgensen, 2021-02-06:
This bug should be fixed in [version 1.21](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.21) so I'll close this

closed 2021-02-06 by KimJorgensen