KimJorgensen/KungFuFlash #84: Kung Fu Flash not working

ref522, 2021-02-04:
I purchased a Kung Fu Flash cartridge from Ebay and it doesn't boot. The led under the reset button keeps flashing. Tested it on a NTSC C64 and SX64. I ordered ST Link V2 programmer to try and re flash it. Does anyone know why the LED keeps flashing and it has a black screen on the monitor.

gooner2077, 2021-02-04:
Have you tried cleaning the contacts on the cartridge as I had this problem with one I built and when I cleaned it it worked.
Also I would contact seller on ebay

KimJorgensen, 2021-02-04:
Yes, please contact the seller.
I have no problems with people selling Kung Fu Flash but then they should also be willing to handle the support if something doesn't work.

ref522, 2021-02-04:
Kim,

 I understand where you are coming from. Thanks for all your work on the product.

Richard

closed 2021-03-31 by KimJorgensen