KimJorgensen/KungFuFlash #88: C128 Mode

worrycan, 2021-04-03:
There is no useful "booting-cartridge" for C128 mode. That's a pretty, because 4,5 mio sellings will ignored. Wouldn't it be wise to offer at least and at last one solution? I want to write a simple BASIC surface in V7 that will appear after start and keeps the C128 mode. I want to use the advantages of the 80 columns resolution. But the KFF switches in C64 mode after turning on a C128. Maybe this could be fixed in further firmeware version? What do you think?

KimJorgensen, 2021-04-04:
Not sure that I fully understand what you want but it is possible to enter C128 mode by pressing F6. If you keep the computer in C128 mode after reset then enable "Persist BASIC selection" in the options menu (F5).

worrycan, 2021-04-04:
I'am thinking about ordering a KFF, so my information is just form reading in the internet. I want to power on my C128 and run automatically a native custom C128 file, without doing anything else than push the power button. But the C128 will switch yet first in to C64 mode, because it's recognizing the firmware as C64 software. Commodore had build in this procedure, when a cartridge appears made for a C64.
Simple question: Does the KFF works completely without using C64 mode? I don't think so... For setting up the KFF I would accept this. But what I'am trying to find is a kind of C128 automatic-bootloader leading to a permanent chosen C128 file by power on the device. The rest would be done though my basic code.

doughamm, 2021-04-06:
Tagging onto this issue: it is not clear from any documentation I could find nor reviews on Youtube what the extent of C128 compatibility is.  Can 128-specific cartridges, particularly WarpSpeed, function natively in 128 mode?

KimJorgensen, 2021-04-06:
Currently C128 support is quite limited; It is possible to disable KFF and enter C128 mode (F6) and ​​there is an option to start EasyFlash emulation that supports 2MHz reads, which is ​needed for Prince of Persia​ and possibly other games on the C128.
There is no support for C128-specific cartridges nor loading of PRGs in C128 mode.

I'll label loading PRGs in C128 mode as an enhancement and would like to keep this issue limited to that. @doughamm Feel free to create a new feature request if you would like support for specific C128 cartridges

worrycan, 2021-04-07:
Oh thanks! Sounds very good you now recognise keeping the C128 mode from switching on to switching off as an issue. Because it is - there is so solution from easy flash over 1541U up to chameleon that provides such a feature. Nothing for native C128 using. Users always forced to use the system as a C64. But the real advantage of this idea is not the 2 mhz mode, but the 80 column resolution. This looks really great and make a C128 interesting and productive even today...

Regards

pmandes, 2022-09-19:
Hi @KimJorgensen 

When I mount a D64 image, it goes into C64 mode immediately.

Would it be technically possible to just mount the image and then switch to C128 mode by pressing F6?

Of course, running binaries in this mode is another matter, but if only it were possible...

[image]

paulhocker, 2023-02-04:
Has anyone had success using a C128 CRT image like the diagnostic cartridge? I have tried BIN, CRT and ROM images and no luck. It keeps telling me that the format is not valid. Thanks.

paulhocker, 2023-02-06:
I believe I figured it out. Looking at the cartridge headers some of mine were not correct. After rerunning the cartconv program it fixed them. Seems that only Generic and Warp128 are supported on C128 at the moment. All other "c64" carts work as expected.

defc0n, 2023-06-15:
It'd be great to have this feature. Realized today that I couldn't load up kipperterm128 via KFF :(

Gun-neR, 2024-01-23:
Wow... I 2nd the last comment.  I just purchased my first C128 and was sorely unimpressed when I realized I can't use KFF to access anything on the KFF SD card while in 128 mode!

Unless I am missing something very important here, I guess it is back to "old skool" ;P SD2IEC while in 128 mode.