KimJorgensen/KungFuFlash #89: Commodore 64C problem

penny-bit, 2021-04-09:
Hi, I've just bought your cartridge from TFW8b and I've inserted it into my **Commodore 64C** and I had many problems: freeze, glitch, not showing the menu etc. then I've inserted it into the good old **Commodore 64** and it worked perfectly fine. 
Here the specs of the motherboards:
- **C64** Assy 250407 Rev A
- **C64C** Assy 250469 Schematic 252311 Rev B

My questions are: 
- Do you know if there is a known issue with this type of C64C/motherboard?
- If yes, Is there any workaround to make it work?
- If no, and the problem is the C64C can you tell me, if you know, what could be the cause of this?

Thank you a lot for your work it's really great!

gooner2077, 2021-04-10:
I have the same c64c as you mine works fine.
Hope this helps.

KimJorgensen, 2021-04-10:
I also have that exact C64C model and it works without any issues and it is in fact that Commodore 64 I use for most of the Kung Fu Flash development.
Are you using the latest firmware release? I did fix some C64C timing issues in v1.18

penny-bit, 2021-04-10:
Thank you for your answer. I've just noticed that my board is **Rev A** instead of Rev B. Sorry about that!
I use the 1.21p is that correct?
Does where is was made has something to do with this issue? Because it's made in Honk Kong, I don't know I'm just guessing.
What could cause the cartridge slot not working properly? 

Here some pics of my C64 e Kung-Fu…
- KFF https://drive.google.com/file/d/1VQXUpJTJ_mN_4w0ZTv_tQKGjCvWveboD/view?usp=sharing
- C64C https://drive.google.com/file/d/1Wtk2_yc1McnqQyye06e9Wg0cuf8QYxUP/view?usp=sharing

gooner2077, 2021-04-10:
Have you tryed cleaning the contacts on the cartridge with isopropyl
alcohol that normally sorts out any problems I have on my old bread bin c64

penny-bit, 2021-04-10:
@gooner2077 but the cartridge is brand new, I've just bought it. I cleaned the contact of the cartridge slot with air.

gooner2077, 2021-04-10:
Like I said it worked for me when mine one was playing up on my bread bin
c64 but worked fine on my c64c.
Was only a suggestion it might work it might not.

penny-bit, 2021-04-10:
Thank you, your suggestions are really appreciated!

I've just tried again, I've been able to load few .crt file and it worked for a while but then, when I press the MENU button, I got this:
- https://drive.google.com/file/d/1r2vlNqBowI3Dlb2CvE8QudPHmtkv-jDU/view?usp=sharing
- https://drive.google.com/file/d/1OLjqwA76v2FKXrIs455l-UBkic_LCGGu/view?usp=sharing
- https://drive.google.com/file/d/1wdBCuhmm2L2xG_ukxwBRN3lAjPe9gjot/view?usp=sharing

Does it helps troubleshooting the problem?

Thank you for your help!

mwedmark, 2021-04-11:
Double check contact points on kff board. I've seen pcb that not been 100%. And different connectors on different computers does not have the exact same width. Depending on wear and tear. In my case the very outpost connector got smaller tears after 10 reseatings.  My quick fix was a thin layer of solderpaste on the outmost Few connection points on pbc. Not sure if it worth it.  Do this on your own risk.

uffejakobsen, 2021-04-14:
I have similar issues with KFF in certain boards - while others are rock-stable - but my issues are always initiated by me pressing a button. 

My KFF has upwards pointing buttons - and when pressing them you inadvertently wriggle the cartridge a little. 
If I stabilize the cartridge by putting a rubber eraser (or something similar with the correct height) below the cartridge - pressing the buttons it safe/working again...

Most cartridges have backwards pointing buttons - pressing them usually does not introduce as much wriggle to the cartridge as upwards pointing buttons does..

@penny-bit Now I know this is a long shot - and it may not explain your problems - can you give some more details on the crashes. freezes and glitches that you see ? do your issues appear without pressing buttons or ?

PS: I've created an feature request for a new PCB revision with backwards pointing buttons #77 - if this proves to be the issue for you then you could consider voting/chiming-in on this issue - that will give a better a picture of how many users that have these kinds of problems

mwedmark, 2021-04-14:
I'm not sure but probably those Cart will still be unstable (just a little better) with those buttons. The real problems can be seen if you put the Cartridge in and out 25 times (scaring the contact points). One or several will have less deep scaring (or nothing at all). Those are the ones that don't touch in your port. If you have the opportunity, try those in another computer they may work there.

penny-bit, 2021-04-18:
I did some more tests yesterday on the C64C ([specs above](https://github.com/KimJorgensen/KungFuFlash/issues/89#issue-854924719))…

After I **cleaned all the contacts** of the cartridge slot with a contacts cleaner spray, bended them a little bit inward, then I’ve also cleaned the contact of the **VIC 8565** and the **SID 8580**, just those two because they are the only ones socketed, then I waited for the spry to get dry.

Then I’ve plugged the KFF and at the beginning it worked fine and I’ve been able to load a Dead Test Cart and it started just fine, but after the first cycle it stopped with the screen I’ve [posted above](https://github.com/KimJorgensen/KungFuFlash/issues/89#issuecomment-817169435). After this I tried to press menu or reset on the KFF and again same freeze screen after a few on and off cycles I was able to start again the Dead Test Cart but again it didn’t even finish the first cycle test.

One thing I’ve noticed in general, if I erase the file **KungFuFlash.dat** file on the SD with another computer sometime (not always) it lets the KFF start again.

After all this, I’ve plugged the KFF into the breadbin C64 ([specs above](https://github.com/KimJorgensen/KungFuFlash/issues/89#issue-854924719)) with the KFF loaded with Dead Test Cart and it did 10 cycles without any problems. I’ve pressed menu and reset buttons without any problems. It works just fine.

Something I can’t explain though is that Dead Test Cart didn’t show any problems on both boards. Other strange thing is if I plug an original **Final Cartridge III** in the C64C it works just fine.

@uffejakobsen Thank you for posting here, but as far as my knowledge goes, I don’t think the problem here is the position of the buttons because if it would be that all the time I press some buttons I would get problems, instead sometimes (especially at the beginning) it works fine.

I really looked for solutions on internet, forums, YT, everywhere  but unfortunately I didn’t find any answers, everybody says to clean the contact, change the cart connector etc. but at this point I actually think the problem is something else, some strange defect on my board which I can’t troubleshoot. If anyone has some other suggestions it’s really appreciated.

I hope all this makes sense and helps.
Thank you all.

zitev, 2021-04-18:
A possible problem could be if the supply circuit is unstable (dried capacitors, low supply voltage, etc.), so the base machine may still be running, but faults will already occur from an extra load.

uffejakobsen, 2021-04-21:
@penny-bit Thanks for your reply
Could you try with another microSD-card ? 
I have seen issues that may relate to the SD card access - I do not fully understand the issue - 
one system turning on with the KFF inserted - but without a SDcard inserter into KFF - then inserting is later seems to make is more stable...
Again this is just my initial observations - it may be wrong - as I need more ting to do experiments before opening an issue

medzes, 2021-05-01:
I've noticed the SD-card stability on one C64 I have too - I'm not sure what that is either. I have experimented with some firmware modifications to reduce the interrupt duration with just 2 or 3 168MHz cycles from the default KFF menu startup code, and that change alone significantly improved the stability of the cartridge + SDCard on that particular C64 I have issue with.
It is indeed hard to pinpoint what exactly the issue is.
Maybe it is just that the PHI2 clock signal on that C64 board of mine has a higher jitter than others, causing the next ARM interrupt to be one or two cycles shorter on average

penny-bit, 2021-05-03:
@medzes how did you modify the firmware in order to reduce the interrupt duration?

Other and final test...
I bought a C64C board on ebay identical as the one I have. It was sold as working board. I just swapped the VIC, SID and CLOCK chips from my board to the new one. It passed 100 cycles of diagnostic cart without any problem but the KFF still have same problem. I don't think the new board is not working, or it's really improbable that two boards have the same identical issue. So the only thing are the chips coming from the original board, but I checked them with the oscilloscope and all the clocks and voltages are OK. So at this point I really didn't know what to do.
After few days, I though it would be a good idea to try to fine tune the DOT CLOCK. I tried to set the small trimmer CT1 close to the clock chip and I noticed that if I rotate it a small amount back and forth trying to find the right setting (I just tried because I don't own a frequencymeter) this would make the KFF more stable, it crashes less. Now I don't know this is right or it's just a luck, but I think all this issue definitely it has something to do with the clock of this particular board.

Thank you all for your support and suggestions.

KimJorgensen, 2021-05-03:
@penny-bit Thank you for your feedback. I have tested Kung Fu Flash on different C64 boards but not on the 250469 Rev A so it is possible that this particular board has stricter timing requirements. However, it would be nice if this bug could be reproduced without swapping chips.
Regarding adjustment of CT1; can't you use the oscilloscope to measure the frequency?

penny-bit, 2021-05-03:
@KimJorgensen Sorry, I didn't express myself well. I meant the new board was delivered without those chips and I had to plug mine into it, I don't have a spare VIC, SID and CLOCK chips to try out.

Apparently the CT1 trimmer adjusts really tiny differences of the crystal oscillator Y1 so those differences are extremely hard to spot on an handheld oscilloscope, at least for me, so I think a frequencymeter would be better for this matter. Moreover, do you think a so tiny difference would make the KFF stop working properly?  

After all this tests I think the board has some weird incompatibility with the KFF or, more probable, the VIC witch is the chip that generate the clock for all the other chips (if I understood right) is somehow faulty and after few minutes, it might get too hot, and it stop to work properly.

Now I just played with the C64C and the KFF and it seems to work fine, it just crashed once, but definitely better than before.

Thank you

medzes, 2021-05-04:
@penny-bit 
The CTO should be able to change the frequency a little (I dont know how much but much less than 1%? I've not tried it). It would have both an effect on the frequency, the startup speed of the crystal and on the clock-jitter, so the stability of the clock delays.

My guess would be a small change to the CTO changes the Q factor of the resonator, which causes the oscillator to have less jitter. And I believe it is the jitter that can occasionally cause shorter periods between ARM interrupt. (so say 5ns jitter late followed by 5ns jitter early could reduce 2 cycles for the arm for one interrupt)

To change the arm interrupt delay, I changed the C code around a little- like swapping around legs of the case statements and if statements. I also dug into the generated assembly and removed some spurious instructions. 
Unfortunately its very hard to predict what will happen when you make such changes, I measured the effect using an oscilloscope.

KimJorgensen, 2021-05-04:
@penny-bit I meant that it would be nice if this could be tested on different 250469 Rev A boards with their original chips to see if it is the revision of the board that causes the issue. Then I could get such board and start debugging the problem.
However, if it is due to the chips or adjustments on the board then it wouldn't help much to acquire that specific board revision.

michielboland, 2021-05-07:
I have a C64C that also has problems with the KFF. It handles other cartridges (UII+ etc) just fine. The KFF also works fine in my other boards.
My issue is mainly that on powerup there appears to be a 50% chance of things getting stuck. For example, if I flash a trivial .prg, or just leave out the SD card, about half of the time the screen will be black, and the other half the program runs fine, or I get the "no sd card" message.
I think I'm on the latest firmware (1.23p)
FWIW my C64C board reads PCB No.252311 REV.4, not sure where this fits in the REV A / REV B timeline.

zitev, 2021-05-27:
I am repairing an Assy 250469 Rev.3 motherboard right now, i am experiencing similar uncertainty with the KFF, at first it was as if the T1 capacitor trimmer was affecting the stability, but i didn't even have time to examine it with an oscilloscope. The Jupier Lander CRT-image also produces strange phenomena, the track image crashes, the music starts and then a few sounds later it completely fades (but otherwise there is a SwimSID in the machine)...

Arcahell, 2021-09-18:
Good my board is the same but mounted on a C64G, the same thing happens to me when pressing the reset button, it stays frozen

What strikes me is that pressing the cartridge down near the buttons when restarting the cartridge with the reset button, the last program or game executed comes out well, if I upload it up the home screen already comes out without loading the last program but with fewer bites than normal,  although I still have the screen when I press the menu button and take the cartridge up or down.

Maybe the expansion port is somewhat unstable but also this board of problems in the kff and that I have updated to version 1.26p

Could someone tell me what could be the problem of the cbm freezing when pressing the menu button?, thanks

zitev, 2021-09-18:
This is a good question, first of all it is worth checking if the T1 trimmer is tuned to the correct frequency (14.31818 MHz on pin 8 of 8701)

KimJorgensen, 2021-09-19:
You can try the new v1.27 firmware where a diagnostic tool has been added that can measure the clock frequency of the C64.

zitev, 2021-09-20:
Hello Kim,
How accurate is this diagnostic tool? I set it up with an oscilloscope, and compared to that, this kind of method shows a little difference, I had to set it up a little later. I don't know which value is closer to the reference. One request: You could enter the two reference values ​​in the tool at the next update so that laymen do not have to search for exact values ​​during calibration. Thanks!..

KimJorgensen, 2021-09-20:
Hi Zitev,

What values did you get?

The accuracy of the tool should be down to the precision of the clock circuit on the Kung Fu Flash board. I did a test with 8 different KFF boards on the same C64 and got slightly different results with the largest deviation being 39 Hz. It should be noted that the frequency of the C64 can also change slightly when getting warm.
This tool was only intended to aid troubleshooting not for laymen doing calibration :)

zitev, 2021-09-20:
Hi,

On the c64 panel, after replacing the T1 trimmer capacitor, I set the PAL clock signal to 14.31818 MHz as measured by a scope on pin 8 of the MOS8701.  KFF's new diagnostic tool showed 985206 Hz (deviation is minimal). True, now I have not checked with a scope how the current state differs from the set one..

Arcahell, 2021-09-21:
Good and thank you very much for this chat, and updated the kkf but the menu does not come out, someone could tell me how to run the measurement tool in usb mode, basic, I know the procedure of the buttons but so far, thank you

zitev, 2021-09-21:
Hello Arcahell,

The T1 trimmer is visible around the center of the motherboard, a tiny, lemon-yellow, adjustable capacity. You can adjust it with a screwdriver that fits exactly into it, by turning it carefully. I think an oscilloscope can be a great help in setting the exact frequency (the scope dipstick shall be measured on the lower left foot 8 of IC 8701 to the left of the T1 trim, the correct frequency for PAL machine is: 14.31818 MHz), but since the machine itself seems stable even with a relatively large frequency deviation, it is not absolutely necessary ...

KimJorgensen, 2021-09-21:
@zitev Are you sure about the 14.31818 MHz? Isn't that for a NTSC model and 17.734475 MHz for PAL?
But anyway, did adjusting the frequency on your C64 make it 100% stable or do you still experience crashes?

@Arcahell For using the USB mode you need to use terminal program on the PC to connect to the serial port that Kung Fu Flash shows up as. I'm using minicom on Linux.
I have not yet tried KFF on a 252311 Rev. A, but on a Rev. 4 and a Rev. B board, and both worked without any issues.

zitev, 2021-09-21:
@Kim: you are right, I did it a long time ago and now due to a topic I searched the net and suddenly found the NTSC master clock. Sorry for the disinformation, so the correct PAL clock is 17.734475 MHz.
I have Rev. B, there is a bit of start-up uncertainty with that, but for now i don't know if the machine itself is unstable anyway, or just with the card - uncertain. One thing is for sure: setting the clock increases / decreases the uncertainty.

KimJorgensen, 2021-09-22:
@zitev OK, thanks for info. It sounds like more investigation is needed to fully understand what causes this instability other than the clock frequency.
Please let me know if others have tried to adjust the C64 clock frequency and what the results were.

Arcahell, 2021-09-22:
> @Arcahell For using the USB mode you need to use terminal program on the PC to connect to the serial port that Kung Fu Flash shows up as. I'm using minicom on Linux.
> I have not yet tried KFF on a 252311 Rev. A, but on a Rev. 4 and a Rev. B board, and both worked without any issues.

I greatly appreciate the information and even more your research, if you have on this page a link of c64 plates that work or present any anomaly for future revisions I would appreciate it, a great greeting

KimJorgensen, 2021-09-22:
@Arcahell I'll update the readme if I find that you can pin problems to a particular board revision. However, I doubt you can do that given how Commodore mixed and matched components when they produced these machines.

Maybe we should compare pictures of our boards to see if there is correlation between the problems and the components used?
For reference, here is my rev. 4 board on which Kung Fu Flash works without any issues: 
[image]
I'm not sure if this came from the factory with out the CT1 trimmer, but instead of a trimmer it has a capacitor soldered on the back of the PCB. When I got the board it came without VIC-II and SID and I had to replace of the CIA chips on this board.

Compared to the rev. A, it seems to use a PLA from a different manufacturer, maybe that is a clue. @penny-bit @zitev and @michielboland do you have a photo of your boards that you can share?

rapiqui, 2021-09-23:
> @Arcahell I'll update the readme if I find that you can pin problems to a particular board revision. However, I doubt you can do that given how Commodore mixed and matched components when they produced these machines.
> 
> Maybe we should compare pictures of our boards to see if there is correlation between the problems and the components used?
> For reference, here is my rev. 4 board on which Kung Fu Flash works without any issues:
> [image]
> I'm not sure if this came from the factory with out the CT1 trimmer, but instead of a trimmer it has a capacitor soldered on the back of the PCB. When I got the board it came without VIC-II and SID and I had to replace one of the CIA chips on this board.
> 
> Compared to the rev. A, it seems to use a PLA from a different manufacturer, maybe that is a clue. @penny-bit @zitev and @michielboland do you have a photo of your boards that you can share?

That's correct. On that Rev.4 the SuperPLA was made by Sharp, whereas on @Arcahell Rev. A was made by Yamaha. 
I have C64Cs with both boards and SuperPLAs and KFF. Let me check out how it behaves in both cases, in case that could be of some help.

zitev, 2021-09-23:
@Kim: I was wrong again, I don't have a Rev B plate, but a Rev 3 plate. Specifically, it looks exactly like in your photo, only CT1 is in place (C64E S/N 086653/87/88). I brought this plate back from the dead, someone has already fixed it and after changing the user port I had to change the user port again, then hunt for soldering errors, replace most of the capacitors, the SID was dead, now it goes with SwinSID, almost it is in all IC sockets, so it's a wonder if it's still working, I'm glad it's just as unstable as it is now - sometimes there is some uncertainty about the KFF, especially at startup - if it's already started, it's stable.

michielboland, 2021-09-23:
> Compared to the rev. A, it seems to use a PLA from a different manufacturer, maybe that is a clue. @penny-bit @zitev and @michielboland do you have a photo of your boards that you can share?

[image]

See above. The PLA reads
251715-01
YM33535
84 08 75 C

I updated the firmware to 1.27p, the diagnostic says
Phi2 frequeny: 985158 Hz
although if I measure the clock with my Rigol DS1054Z it reads 985.253kHz, which is closer to what it should be.

[image]

The KFF still fails at poweron more often than not on this board.

KimJorgensen, 2021-09-23:
> That's correct. On that Rev.4 the SuperPLA was made by Sharp, whereas on @Arcahell Rev. A was made by Yamaha.
> I have C64Cs with both boards and SuperPLAs and KFF. Let me check out how it behaves in both cases, in case that could be of some help.

Interesting, thank you for the details.
 
> Update: I have been able to test the KFF on two C64c, one with 252311 rev. 3 with Yamaha SuperPLA and another with 252311 rev.4 with Sharp SuperPLA:
> 
> Kung Fu Flash firmware version: 1.23b
> 
> Couldn't replicate the mentioned behavior. Apologies.

No need apologize, it just shows that you cannot use the C64 board revision or PLA model as an indicator to which boards will not work reliable with Kung Fu Flash. Unfortunately, it also makes my job of debugging the problem harder as I don't own a C64 that has problems with KFF and I don't know what to look for if I would like to acquire one that does.
@rapiqui  Thank you for testing this.

> I updated the firmware to 1.27p, the diagnostic says
> Phi2 frequeny: 985158 Hz
> although if I measure the clock with my Rigol DS1054Z it reads 985.253kHz, which is closer to what it should be.

@michielboland I would trust the Rigol. As I wrote earlier, the accuracy is down to the precision of the clock circuit on the Kung Fu Flash board and that hasn't been calibrated to provide a super accurate frequency (nor should that be necessary). 

Thank you for testing and providing measurements

penny-bit, 2021-09-23:
> @Arcahell I'll update the readme if I find that you can pin problems to a particular board revision. However, I doubt you can do that given how Commodore mixed and matched components when they [...]

Sorry for the late answer but I'm really busy at work. I hope it still useful. Here my board:
https://drive.google.com/file/d/1Wtk2_yc1McnqQyye06e9Wg0cuf8QYxUP/view?usp=sharing

I didn't have time to read all the recent comments, I hope to get through them in the next days.

Thank you all.

KimJorgensen, 2021-09-25:
Maybe we should try something else. As it seems that most problems occur at startup or when entering the menu it could be that there is a problem with resetting the C64 properly from the Kung Fu Flash.

I have made and attached a test firmware where the reset line is held low for longer. Please let me know if it makes any difference.
[KungFuFlash_vTST1.zip](https://github.com/KimJorgensen/KungFuFlash/files/7230088/KungFuFlash_vTST1.zip)

lordoftears, 2021-09-26:
My KungFuFlash works with every C64 I have, but I have a curious issue with my Sixty Clone: often (3 on 5 attempts), when I press the Menu button, I have garbage on screen. Then I press it again and everything is fine. Games are loaded fine and I can enjoy the cart.

I performed the Diagnostic due to the new 1.27 firmware. With the Sixty Clone the clock is moving from 887450 to 987750, while it is ok (fixed value) on a breadbin C64 I tested.
I have no other issues with the Sixty Clone, which is usually connected to a Ultimate 2+ device and has a good video on my 1084s monitor.

michielboland, 2021-09-27:
I tried the TST1 firmware - not much improvement.
I made a video of running an Ultimax crt while repeatedly turning the machine off and on. The cartridge should display a horizontal color bar pattern (using color RAM) on grey background and border. Sometimes the color pattern would be visible, which would indicate to me that the cartridge has run, but the background or border color would be wrong, or there was other garbage in the display.
https://youtu.be/fWGGY_EXSLQ

KimJorgensen, 2021-09-27:
@michielboland Thank you for testing the firmware and for the video. I guess we have to look at clock signal or power supply if it is not the reset signal that is the problem.
When you had the scope out, did you capture the clock signal during power up? It could be interesting to see if it is stable when the KFF starts to run.

Do you only have problems during power on, and not when pressing menu like some of the others?

michielboland, 2021-09-29:
I get a clock signal at poweron and then nothing happens for about 250ms (I guess reset is being held low during that time.)

KimJorgensen, 2021-09-29:
OK thank you, that will be helpful :+1:
Could you also be persuaded to have a look at power supply during power up, if you cannot detect anything strange with the clock signal?

lordoftears, 2021-09-29:
Yes, I can do that. You mean activating the diagnostic just after the power up?

Emanuele

KimJorgensen, 2021-09-29:
@lordoftears Sorry, I should have tagged @michielboland :)

lordoftears, 2021-09-29:
No problem :D

Emanuele

michielboland, 2021-10-06:
+5V rail broken:

[image]

+5V working:

[image]

(Cyan trace (probe 2) is R/W)

There appears to be some inexplicable voltage drop - not sure what's going on there.
Also I noticed there is 10 Ohm between GND and +5V with the machine unpowered, surely that can't be right?

KimJorgensen, 2021-10-06:
@michielboland Thank you for measuring this. I don't have a digital storage scope so I can't compare this to another C64. However, I'm not sure that this is a problem if the reset is held low until the power and clock is stabilized.

Was reset released after 250ms after power on? And here the R/W are looking OK (but noisy) after about 1.5ms if I'm reading that correctly?

KimJorgensen, 2021-10-13:
@zitev KFF is already checking for a stable clock before releasing the reset line. Maybe that detection isn't good enough or maybe the fault has nothing to do with the clock at all. At this point it is all pure guesswork.

michielboland, 2021-10-13:
These two happened to be ok. Captures look the same for failed startups.
Here is a capture I managed to make of some noise that appears on when power is initially applied (i.e. before there is a clock)
(this time the machine failed to boot)
[image]

michielboland, 2021-10-16:
The PAL upd file in the TST2.zip appears to be the same as in the TST1.zip

KimJorgensen, 2021-10-16:
@mwedmark OK, not that either :( but thanks for testing

michielboland, 2021-10-17:
No improvement I'm afraid.
I have observed some different failure modes, FWIW. Sometimes the machine will start up fine. Other times the machine crashes instantly. Sometimes the R/W line shows 3 W pulses in succession almost right after the reset line goes high which would mean the CPU is executing a BRK (0) instruction very early on after powerup (?).
Another failure mode is that the machine will run fine for what appears to be some milliseconds but would then lock up anyway.

I'm using the following image for testing
[test5.zip](https://github.com/KimJorgensen/KungFuFlash/files/7359903/test5.zip)
source below

```
; Ultimax

* = $e000

nmi_handler
  rti
rst_handler
  ldx #0
  ldy #8
loop
  .rept 62
  sty $d020
  stx $d020
  .endr
  bit 0
  nop
  jmp loop

* = $fffa

  .word nmi_handler
  .word rst_handler
  .word 0
```

michielboland, 2021-12-28:
Problem appears to be fixed in 1.32. I couldn't reproduce a hang with about 10 poweroff/powerons; when I revert to previous firmware it will hang randomly.

KimJorgensen, 2021-12-30:
@michielboland Great that it works :)
I guess the code optimization must have changed the timing, but it would have been nice to know where the problem was so I do not make it worse again by mistake

michielboland, 2022-01-01:
Firmware
v1.31 : boot fails intermittently
v1.32 : boot ok (tried about 20 times)
Hope that narrows it down a bit.

TzOk83, 2022-02-02:
I have similar problems (garbage on a screen) when trying to enter KFF Menu out of a Dead Test or a Diagnostic Cartridge. Other cartridges seem to work fine. I have PAL C64C machine, and KFF v1.34.

lordoftears, 2022-05-12:
The 1.39 firmware update fixed my garbage screen issue when pressing the menu button. I also noticed that the diagnostics frequency is more stable now on that computer.

[16 more replies]