KimJorgensen/KungFuFlash #90: CRT and PRG to run together

Gabeki17, 2021-04-23:
Is it possible for example to have Epyx fastload crt active whine runing a prg automaticaly also?
Is there a possibility already I missed to find out about might be?

modeler, 2021-04-28:
I wondered this also, I wanted to have KFF act as an Action Replay cartridge, then load a PRG file from the SD card and disassemble it with using the emulated AR. I don't think it's possible, you'd need two KFFs and a cartridge port expander board assuming that would work?

Gabeki17, 2021-05-03:
If the prg is on a floppy or a id2iec only a command has to be run. Even 10 kepress can be loaded into keyboard buffer. (even the return key) so it would simply run. ( It is a dirty workarund, but I am actually using. Like I have a loader for CBM Command. My loader is simply pokeing key "i" to change the view, then another key is also poked to go a level up to root dir. so after CBC Command have started the key presses are taking effect. it is zero touch.)

Gabeki17, 2021-05-13:
In the end i have changed Epyx Fastload crt to run a prg after it have loaded itself.

KimJorgensen, 2024-12-29:
Closed as a duplicate of #20

closed 2024-12-29 by KimJorgensen