KimJorgensen/KungFuFlash #93: Alternatives for stm32f405rgtX

par00s, 2021-05-14:
Hi,
Lastly I found that Ali link for stm32f405rgtX is outdated and no other retailer has this chip. All are out of stock.
Even if someone have active offer for it, after a few days after ordering, everyone notifies you that they no longer have the chip and asks you to cancel the order.
There are some alternatives to this chip or no?

KimJorgensen, 2021-05-14:
There is a global shortage of STM32 chips so it seems that these are hard to get from anywhere. I assume that a STM32F415RG can be used as an alternative but this is not something that I have tested.

Sgw32, 2021-05-17:
F for STM32 :( 
I've bought F405 for 3-5 USD before and now I pay 13-20 for it. 

Some people say it will last for 9 months, others think about 2-3 years unfortunately.

ogg1e, 2021-05-17:
You can get them on ebay and aliexpress, but you run the risk of fakes or used chips.

par00s, 2021-05-17:
@KimJorgensen Thanks. I will check this chip and let you know :)

@ogg1e Kepp in mind that I told before about availability on Aliexpress. I just didn't check availability on the most expensive offerings, above $20 a pop.

ogg1e, 2021-05-19:
you could buy dev boards off ebay for around $11 USD and remove the chips from them.

msolajic, 2021-06-22:
Did anyone try this instead of the original STM part? https://lcsc.com/product-detail/GigaDevice_GigaDevice-Semicon-Beijing-GD32F405RGT6_C99031.html

Sgw32, 2021-06-22:
> Did anyone try this instead of the original STM part? https://lcsc.com/product-detail/GigaDevice_GigaDevice-Semicon-Beijing-GD32F405RGT6_C99031.html

The original part is available for the ~same price or lower(in certain countries)

StormTrooper, 2021-07-01:
These devices are now almost impossible to get hold of without paying crazy prices. Lead times now of April 2022 :(

ogg1e, 2021-07-01:
Like I said above, I bought a few dev boards off ebay and removed the chips from them. Works fine. I have two for sale on ebay, but no one seems interested. https://www.ebay.ca/itm/174803233006

par00s, 2021-07-12:
> There is a global shortage of STM32 chips so it seems that these are hard to get from anywhere. I assume that a STM32F415RG can be used as an alternative but this is not something that I have tested.

Big thanks for this advice! :)
Luckily, I was able to buy some STM32F415RG chips on Aliexpress for the price of ~8$

I used and tested them yesterday. Works like a charm.

repetto74, 2022-04-30:
> Did anyone try this instead of the original STM part? https://lcsc.com/product-detail/GigaDevice_GigaDevice-Semicon-Beijing-GD32F405RGT6_C99031.html

I have tested the KFF with this alternative version but I cannot get into DFU mode to program it. Does anyone succeeded to program this clone in DFU mode and how to do that?

repetto74, 2022-04-30:
I have taken a GD32 clone from aliexpress hoping this was a valid replacement but I cannot get it to DFU mode while trying to program it via USB and the Dfuse software. Does anyone used this GD32F405RTG6 for the Kung-Fu Flash and got it working?

CqaZ80, 2022-06-30:
> Did anyone try this instead of the original STM part? https://lcsc.com/product-detail/GigaDevice_GigaDevice-Semicon-Beijing-GD32F405RGT6_C99031.html

I got a GD32 too from ali, I got programmed it with the same cheap programmer clone STM32 from aliexpress that I use for original STM32 parts, furthermore, the software detects it as the original part with no issues when it gets programmed and without warnings everything ok till here, but the GD32  refuses to work as is expected I mean it doesn't work at all, totally dead... I read somewhere that GD32 has some differences with STM32F405RGT6 something related to internal timers... It seems GD32 is not 100% compatible with STM32... It's a pitty

Sgw32, 2022-07-01:
> I got a GD32 too from ali, I got programmed it with the same cheap programmer clone STM32 from aliexpress that I use for original STM32 parts, furthermore, the software detects it as the original [...]

Yes, but I plan making/fixing GD32 version anyway. It will help to make cheaper and easier KungFuFlash cartridges. 
As I've said I have some progress, maybe it will work soon.

CqaZ80, 2022-07-01:
good news then!!! are you planning sharing it???

Sgw32, 2022-07-01:
Yes, of course, but it moves very slowly.

pigravity, 2022-08-14:
> Yes, of course, but it moves very slowly.

Hope it will work, bought few of GD32, testing them a bit. Seems like they are quite fine chips.

closed 2024-12-29 by KimJorgensen