KimJorgensen/KungFuFlash #95: C128 diagnostic cartridge support

phako, 2021-05-22:
Currently it seems not to be possible to use KFF as a C128 diagnostic cartridge. This is probably somewhat similar to #88

KimJorgensen, 2021-08-18:
Support for C128 normal cartridges has been included in v1.25

closed 2021-08-18 by KimJorgensen

sdotolo71, 2021-11-08:
Is is possible to have more instrunctions on how this kind of support is achieved since v1.25? i updated to the latest version but not being able to load from KFF some crt files (diagnostic cart for now) expressely built for C128 mode. What do you means for "normal cartbridges" for C128? is possible to have some crt working examples for C128? any help will be appreciated for C128 owners because until now the KFF is working pefectly for the C64 machines only!

phako, 2021-11-08:
"normal" C128 cartdriges differ from C64 cartridges such that the identification string of the ROM and the state of the game and exrom pins is different to all combinations the C64 supports (I think high on both)

That's what the code implements.  TBH I haven't managed to test that yet

KimJorgensen, 2021-11-08:
Only bin and rom files are supported, not crt. You can find the C128 diagnostic roms [here](http://blog.worldofjani.com/?p=164)

sdotolo71, 2021-11-09:
> Only bin and rom files are supported, not crt. You can find the C128 diagnostic roms [here](http://blog.worldofjani.com/?p=164)

Thanks for reply Kim and let's say thanks for your optimal work on this topic for we that love the old-8-bit computers. Said that, if i understood well..you are saying that i can put a .bin or .rom file on my SD card (in Kung Fu Flash card) and choose like other files (.d64, .prg, .crt)? if so...this is a good new for me and i'll try and let you know. Thanks anyway

paulhocker, 2023-02-04:
@sdotolo71 were you able to test and see if this worked? I am having trouble with my own cartridge files that run okay on vice, but not on KFF.