KimJorgensen/KungFuFlash #98: Feature request - Port to Vic-20?

beamrider66, 2021-08-12:
Would like to see a Vic version of this, not sure how feasible this is?

KimJorgensen, 2021-08-13:
That should be feasible. You would obvious need to design a new PCB or use an adapter for the VIC-20, and make a lot of changes to the firmware.

pmandes, 2021-08-20:
Hi guys,

I was thinking about the same and did something like this:

[image]

[image]

This is the PCB layout proposal and the pinout of connecting the expansion port with the MCU.

I am open to any suggestions.

Of course, the firmware and the ROM with the file manager are to be rewritten.

ARM could also emulate memory expansion, and its internal FLASH could store cartridge images in slots, e.g. diagnostic, etc.

KimJorgensen, 2021-08-20:
Compared to Kung Fu Flash, you would need to read 3 ports instead of 2 to decode the bus cycle. So for efficiency I would have placed the R/W signal(s) together with the other bus signals and maybe chosen a STM32F407 in a LQFP100 package.

You are of course free to do whatever you like, but if you start your own project I would appreciate if it was called something else to differentiate it from this project.

beamrider66, 2021-08-20:
Great that's good progress.

I would be willing to port the Menu software if we can get someone to work on porting the firmware.

Perhaps we may call this one Karate Flash?

pmandes, 2021-08-20:
Tim, thank you very much for the suggestion about the signals. Changing the package to LQFP100 will obviously entail an increase in cost. 

In fact, I don't want to create a whole new separate project. I rather treat my work as a "proof of concept" that it is feasible. I really don't want to create a competing project if you are going to create a version for VIC20 within the KFF project. I would be happy to participate in such a venture.

KimJorgensen, 2021-08-21:
The price difference for changing the package is quite small, but if cost is a concern I would try to go for a cheaper MCU with less flash. All this is a bit theoretical, with the current global chip shortage in mind.

At the moment there are no plans for a VIC20 version, only this feature request. If there were plans, I would probably have created a separate repository anyway, considering the amount of changes required.
So if this is an area of interest, I would suggest that you just go with it and use whatever you want from this project.

zitev, 2021-12-18:
If porting is already the topic: how easy (or problematic) would it be to port KFF to Plus4?

KimJorgensen, 2024-12-29:
Nobody seems to be willing to implement this. Closing the issue

closed 2024-12-29 by KimJorgensen