KimJorgensen/KungFuFlash #101: Add support to Drean cartridge

beatmaster13, 2021-09-11:
Hi there! Could you add support to Drean's Argentine-made cartridges?

http://www.compuclasico.com/site/user/pages/02.museo/01.commodore/g_gm_dc_02.jpg

You already have VICE support

https://sourceforge.net/p/vice-emu/code/39076/

https://sourceforge.net/p/vice-emu/code/40690/

ID:74

KimJorgensen, 2021-09-12:
Hi,
It seems that the cartridge format is similar to Prophet64 and that it also can be used for emulate the Drean ones.
Do you know if a schematic of the cartridge exists or where to find more ROMs to test this?
/Kim

beatmaster13, 2021-09-12:
Hi, Kim! Thanks for taking my inquiry

Everything I have

https://send.cm/d/4mHs

KimJorgensen, 2021-09-12:
Thanks. Support for Drean cartridges has been added to relase v1.26

closed 2021-09-12 by KimJorgensen

beatmaster13, 2021-09-12:
Thank you very much for your support.