KimJorgensen/KungFuFlash #102: Unstable clock values with Diagnostic feature on 1.27

lordoftears, 2021-09-27:
My KungFuFlash works with every C64 I have, but I have a curious issue with my Sixty Clone (assy 250466): often (3 on 5 attempts), when I press the Menu button, I have garbage on screen. Then I press it again and everything is fine. Games are loaded fine and I can enjoy the cart.

I performed the Diagnostic due to the new 1.27 firmware. With the Sixty Clone the clock is moving from 887450 to 987750, while it is ok (fixed value) on a breadbin C64 I tested.
I have no other issues with the Sixty Clone, which is usually connected to a Ultimate 2+ device and has a good video on my 1084s monitor.

KimJorgensen, 2021-09-27:
Is the clock always unstable when you enter the diagnostic? and does it get stable after running the diagnostic for a while?

What chips have you used to assemble the Sixty Clone board? Are they all original Commodore, or has some replacement chips been used?

lordoftears, 2021-09-27:
The clock is always unstable. At each cycle, it changes, I waited for several cycles but it continues moving. It looks strange because with that oscillation the video output should be bad, instead it is good.
 
The Sixty Clone has original chips except for the PLA and an ARMSID.
Some months ago, when I experienced the issue with the menu button, I tried swapping the chips with a different C64, which mounts original chips, but nothing changed.

KimJorgensen, 2021-09-27:
Yes, I would also expect the video output to be affected if the frequency did change that much. Maybe it is a bad connection on the expansion port?
Have you tried to measure the PHI2 signal with a logic analyzer or oscilloscope?

lordoftears, 2021-09-27:
Unfortunately, I do not have a logic analyzer or oscilloscope, sorry, I cannot be useful in this.
When I encountered the problem at the beginning, I checked with the tester all the connections of the expansion port. The voltage is correct, also other paths seem to be ok. The Ultimate-2+ cart works well, as well as "normal" cartridges.
Also the KFF works, I just have this issue when pressing the menu button, not each time, but often. I also checked the KFF with other computer, it works fine.

ogg1e, 2021-09-27:
I was seeing the same issue on my c64c.  I think it's the expansion port, really. Pushing down on those buttons moves the cart enough to cause things to go bad.   Either we need a new case design so that it has legs that allow the cart to rest on the desk to take the pressure off the port, or KFF needs to have the buttons moved to the backside instead of the top.

KimJorgensen, 2021-09-27:
@lordoftears I don't know about the Ultimate-2+ but most "normal" cartridges do not use the PHI2 signal. It is hard to get to the bottom of this without the ability to measure the signal.

@ogg1e That could explain a problem with the menu button but hardly why the frequency would be unstable when not touching the cartridge.

lordoftears, 2021-09-28:
This evening (CEST Time) I will try on an original C64c, both the menu button that the clock issue.

lordoftears, 2021-09-28:
As I supposed, it works like a charm on my C64c. Frequency fixed at 985214, no menu button issue.

KimJorgensen, 2021-09-28:
@lordoftears Great that it worked on your C64C. Did you use the same power supply for these tests?

Just for info; some have problems with the C64C (see https://github.com/KimJorgensen/KungFuFlash/issues/89) while others have problems with the expansion port connection when pushing the buttons as @ogg1e wrote (see https://github.com/KimJorgensen/KungFuFlash/issues/77)

lordoftears, 2021-09-29:
@KimJorgensen, thank you for the infos. I've used an angled adaptor for the expansion port, because at the beginning I thought it could be the problem. I am following also the other threads.
The PSU are different: a Keelog for the Sixty Clone and an old one of a C64c for the other original C64s (the breadbin and the C64c).

I must try with the same PSU. Anyway, the button menu issue on the Sixty Clone was experienced also with the other PSU.

KimJorgensen, 2021-09-29:
@lordoftears OK, thanks for info.
Regarding the PSU, I was trying to guess why the measured frequency would be wrong and was speculating if that due to noise on the clock signal coming from the power supply. But since you already tried another PSU with the Sixty Clone it can't be that.

However, looking at the service manual for the C64 I can see that the 5V supply for the clock circuit actually is regulated by VR2 on the main board. Do you have a regular 7805 installed as VR2 on the Sixty Clone or are you using some replacement?

lordoftears, 2021-09-29:
I tried anyway with the other PSU, same result.
I made a video about the clock diagnostic to show you the "oscillation". [https://we.tl/t-5slW4hTJW0]

I used a regular 7805 (I took the parts kit from Run Stop Re-Store as I think most of the people having built a SixtyClone).
[https://www.dropbox.com/s/1mjhd0qh7wdb46t/IMG_20210601_222406154_HDR.jpg?dl=0](url)

KimJorgensen, 2021-09-29:
Thank you for the video and picture.
When I read that the frequency was moving between 887450 to 987750 Hz it sounded much more dramatic than shown in the video, where the frequency was moving between 989076 and 989219 Hz.

This had me wondering if this oscillation just is a artifact of the clock circuit being misaligned. Do you know if the clock frequency was adjusted after the board was assembled?

lordoftears, 2021-09-29:
@KimJorgensen I am really sorry, I mistyped the first digit of course... 😕

When I assembled the board, I had black and white image on my 1084s, I adjusted a bit the trimmer and it displayed correctly.

KimJorgensen, 2021-09-29:
@lordoftears Don't worry about it :)

I don't know how accurate the frequency needs to be for the 1084, so I'm not sure what that implies.
Unless somebody else has some insights, I don't think we can get bottom of this unless you somehow measure the Phi2 signal to tell if the frequency is indeed unstable or if KFF has trouble detecting the signal.

lordoftears, 2021-09-29:
I have a multimeter capable of measuring clocks, but I don't think it is accurate at the precision needed. Unfortunately, I cannot afford  an oscilloscope at the moment, I mostly a game player, but if I can find some friend here having that I'll check it. Anyway, I'll try with the multimeter in any case. Apart from this issue on this specific computer, the KFF is something incredible for a game player like me.

KimJorgensen, 2021-09-29:
Glad that you like it.
If there is a bug I'll be happy to fix it, but unfortunately it is really hard to do remote debugging

lordoftears, 2021-10-04:
> 
> 
> Glad that you like it. If there is a bug I'll be happy to fix it, but unfortunately it is really hard to do remote debugging

I know, I hope to be helpful soon, I think everybody here appreciates the great work you did.

lordoftears, 2021-11-15:
@KimJorgensen I was able to perform some tests:

- replacing the new PLA with an original one, I have the same issue
- with the new firmware, it moves a bit in the rage of 986000Hz and 986200Hz or something like that. Less than before.
- Unfortunately I don't have a sharper tool with more digits, but the clock measured with my multimeter is fixed at 985. 2 MHz

Gabeki17, 2021-11-26:
Hi,

I have one device where the clock is 985119. And it is terrible. 

There is a game that might be is the best to test it out.
It is Jupiter lander CRT version.

It is not only sparkling, but the sparkles are even coliding with the lander. :)
Also it goes even worst if warming up. So you die almost instantly. 

Is Kungfu checks clock regularly so it can autotune it? Or only on bo

https://user-images.githubusercontent.com/4869862/143604921-fdae42e6-afe1-4e9f-be6c-f4fe75c9995c.mp4

oting?

[image]

KimJorgensen, 2021-11-28:
@Gabeki17 You could try adjusting the frequency to around 985248 Hz but I doubt this will do much difference. As I already told you in #94; It is probably the KFF that can't meet the timing requirements of the VIC-II on your specific C64.

You could try Jupiter Lander V2 in [this release](https://csdb.dk/release/?id=210760) and see if it works any better. If it does, maybe I can improve the Ultimax mode for normal cartridges as I did for Easyflash. Remember to start the EF cartridge in VIC-II/C128 mode via shift+return.

I would like to keep each issue to one topic so if your clock values is not unstable please keep the discussion to #94

gletonai, 2021-11-29:
My issues on ALDI C64 stopped manifesting with 1.29p, god job.

Gabeki17, 2021-11-30:
> @Gabeki17 You could try adjusting the frequency to around 985248 Hz but I doubt this will do much difference. As I already told you in #94; It is probably the KFF that can't meet the timing [...]

Thx!,

It also effects other games, like Prince of Persia. (Actually I bought KFF because of. :( )

I would try to tune it, but can't compile the code.
If you can help me with that I could try to solve my own issue, and when it works I can simply give you the information about.

I know there are many C64 behaving a litlle different.

Thank you in advance!

With regards,
Gábor

KimJorgensen, 2024-12-29:
I assume that the original issue was due to a timing problem that can be fixed by adjusting the phi2 offset. Otherwise feel free to create a new issue

closed 2024-12-29 by KimJorgensen