KimJorgensen/KungFuFlash #106: IDEA: Working&tested STM32F405ZGT6 version

Sgw32, 2021-12-07:
Hi,

I know it may be useless but I have a working and tested version of Kung Fu Flash with STM32F405ZGT6 , due to shortage it is available on some old devices. 
I want to share the files, but how to contribute(I can create a pull request but how to organize repo structure?)

Here is how it looks:
https://www.instagram.com/p/CWE195NIjbZ/

Best regards,
Fedor.

KimJorgensen, 2021-12-07:
Hi Fedor,

I'm be happy to add a link to your GitHub project for people who are interested in building a version with the STM32F405ZGT6. However, I prefer not to introduce new variants of the Kung Fu Flash hardware due to the additional support that may arise from this.

Best regards,
Kim

Sgw32, 2021-12-07:
I can submit pull request or I'll upload a fork & merge my local repo to github. I think second is preferred. 
Do not worry about compatibility, ZGT6 is just a larger RGT6 package.

closed 2024-12-29 by KimJorgensen