KimJorgensen/KungFuFlash #109: Feature request: Sending CRT-files over USB

mwedmark, 2022-02-04:
I just tried to send an CRT-file over USB and does not seem to work, would that be possible to add?

KimJorgensen, 2022-02-06:
What do you want to achieve by this? Just burn CRT files to flash via USB or transfer files to the SD card?

mwedmark, 2022-02-06:
The reason for asking because I've just got a improvement suggestion for Assembly64 approved and implemented. Not sure if you seen it.  It is a mac/win/Java application as client and a scraper server in cloud which automatically keeps track of c64 releases. 
The improvement I suggested means from v4.3, with some setup of ef3utils, you can just click on a c64 program/one file demo in the tree and it is downloaded and started on C64 using KFF.  It would be very nice to be able to all their crt files the way to test them before putting them onto SD card. But sounds nice,  it there a way to burn crt and start that already via USB? I didn't know that,  please share so I can try and see if it accomplished the same thing.

KimJorgensen, 2022-02-06:
Currently there is no way to burn and start a CRT file via USB. You can only start a PRG file via ef3utils