KimJorgensen/KungFuFlash #110: KFF stuck at loading the same ROM (no menu)

germanviscuso, 2022-02-04:
I was using KFF normally and loaded a program to test joystick ports. As usual, this program loads automatically but now the menu button just loads the same program.

I tried placing an empty microsd card but when I do that KFF enters a loop trying to load the missing program (menu doesn't work either).

Is this now a corrupted KFF? Can it be reset?
Interested to find out what triggered this behavior.

KimJorgensen, 2022-02-06:
From your description, I have no idea if the flash is corrupted or what caused this behavior. However, if possible I would try the KFF with a different C64 to see if it does the same.

Should the flash be corrupt, it is possible to reinstall the firmware with the methods described here: https://github.com/KimJorgensen/KungFuFlash/blob/master/firmware/README.md

germanviscuso, 2022-02-08:
Thank you Kim. Closing this ticket as I will try to reflash the KFF (might have been corrupted)

closed 2022-02-08 by germanviscuso