KimJorgensen/KungFuFlash #111: Feature Request: Load .CRT and use D64

jperrin72, 2022-02-05:
Hi Kim,
First I want to say that your cartridge is abvsolutly amazing :)
I was wondering if you plan at some point to develop support for both using a .CRT and .D64 images at the same time.

1. mount .D64 image
2. select Freeze Cartridge .CRT
3. load prg from D64 image and use cartridge built-in freeze and monitor

I'm curious to know if there are any limitation with the hardware that would prevent to do this.

Thanks

Jerome

KimJorgensen, 2022-02-06:
Yes there are limitations, see #20.
 I'll close this as a duplicate

closed 2022-02-06 by KimJorgensen

jperrin72, 2022-02-06:
thanks for the answer - I was expecting this answer but had to ask :) I just ordered a pi1541 to combine both hardware.