KimJorgensen/KungFuFlash #114: WarpSpeed 128 stars in C64 mode

matti157, 2022-02-23:
KFF 1.35
PAL C128
.CRT from https://rr.pokefinder.org/wiki/WarpSpeed

I can't figure out how to use the KFF in 128 mode. I liked the WARP cartridge to have the wedge also in the 128, but every time it boots in C64 mode.

If I go to the options menu and press F6 for 128 mode the screen flashes as if it has taken over, but nothing changes.

How do I use the KFF in C128 mode?

Thanks

KimJorgensen, 2022-02-24:
You can use the latest version of VICE / cartconv to create a C128 CRT file:
cartconv -t ws128 -i WarpSpeed_v2_1987.bin -o WarpSpeed_v2_1987.crt

matti157, 2022-02-25:
> You can use the latest version of VICE / cartconv to create a C128 CRT file:
> 
> cartconv -t ws128 -i WarpSpeed_v2_1987.bin -o WarpSpeed_v2_1987.crt

Thank you! I will try as soon as possible

matti157, 2022-03-12:
It works, thanks

closed 2022-03-12 by matti157