KimJorgensen/KungFuFlash #116: C64 Game System mode

mwedmark, 2022-03-24:
Suggestion for a new function:
- A settting under Kung Fu Flash settings called: "C64 Game System Mode" which shows an on-screen keyboard when doing search instead of waiting for key input.

KimJorgensen, 2022-03-25:
Good idea. I guess this also requires the possibility to enter the settings menu and the search bar using the joystick

KimJorgensen, 2022-05-27:
@mwedmark [FW 1.40](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.40) now allows you to enter a search string with the joystick. Not exactly what you suggested, but I hope it will still be useful

mwedmark, 2022-05-27:
From my point of view, you just nailed it! Perfect solution, even better than on-screen keyboard. Thanks!
I'm closing the issue as it is resolved.

closed 2022-05-27 by mwedmark