KimJorgensen/KungFuFlash #117: all three LED come on and stay on, black screen

AdrianPilko, 2022-03-26:
all 3 button LED come on and stay on and menu browser doesn't load, just black screen, it did same on two different C64: the cartridge was brand new and with supplied SD card. Couldn't find anyone having same problem after a while searching google, I have emailed the seller. Also briefly tried looking at the source code here. Has anyone else had this problem? I don't expect support from here but thought it was worth a try.

If it's just a case "of it does this when the firmware is missing or corrupt", then that's fine I found instructions to reflash the firmware using dfu_util.

KimJorgensen, 2022-03-26:
It could be problem with the firmware or a hardware related issue.
I would rather not suggest anything that could void your warranty until you have an agreement with the seller

acj1971, 2022-03-27:
Im getting the same issue

AdrianPilko, 2022-03-27:
Also worth saying that the same thing happened with or without an sd card inserted

agodng, 2022-04-15:
Hi at all, this is my first write and I specify that I would like to thank Kim Jorgensen
i got a kungfuflash cartridge from tfw8b version 1.2 (without usb interface) it always worked perfectly after all firmware updates from 1.29 to 1.34, unfortunately the latest update version 1.25 an after 1.37 something went wrong
all the LEDs turned on and the c64 keyboard commands did not respond, very slow

I tried to put only the fw1.37.upg file on the microsd but when I tried to flash the firmware 1.37 it remains in black screen

now I bought a STM32 ST-Link to reprogram the eprom and I hope to solve, I will let you know if everything works with 1.37 through SWD

closed 2024-12-29 by KimJorgensen