KimJorgensen/KungFuFlash #118: Could we release Mac version of the EasyFlash tools?

mwedmark, 2022-03-27:
A guy asked if he could use these tools on his Mac? Are there any Mac-users that could help us out?

fortiko, 2023-07-29:
@mwedmark I have added some very small changes that allow the EasyFlash EF tools to be compiled on current versions of macOS and Apple Silicon that @KimJorgensen has incorporate in the main branch now: https://github.com/KimJorgensen/KungFuFlash/issues/165

mwedmark, 2023-07-30:
Thanks for fixing and sharing!

fortiko, 2023-08-10:
@KimJorgensen @mwedmark I think with the fixes in https://github.com/KimJorgensen/KungFuFlash/issues/165 we can close this as well? Not sure about providing binaries?

magwed, 2023-08-10:
Yes I think you're correct, from my point of view that fixes it. Sorry to say, I'm not a Mac-person myself so this was mostly conveyed from friends/other users of Kung Fu Flash and probably not help with providing binaries.

closed 2024-12-29 by KimJorgensen