KimJorgensen/KungFuFlash #119: Some Issues with v1.36 firmware

zitev, 2022-03-30:
Hi @KimJorgensen,

I ran a ride with the new firmware today. I watched two critical games, one of which is Pharaoh’s Curse, whose title music, along with the old firmware, was played at an accelerated rate, probably due to poor timing, and I couldn’t even enter the game. This error did not occur with the new firmware, everything seemed to be fine...
The other example program is the CRT version of Jupiter Lander, which has worked well for me so far, but now the phenomenon presented by [@Gabeki17 (Sparkling Screen)](https://github.com/KimJorgensen/KungFuFlash/issues/94) has also occurred to me. With the standalone PRG version of the Jupiter Lander, I didn’t experience this under any of the firmware. After restoring the previous firmware version, everything returned to the previous state (although I still see minor flashes on the Jupiter Lander sometimes, but it may have been there so far, I just didn't notice it.)

Thanks for everything,
zitev

KimJorgensen, 2022-03-30:
@zitev Thanks for testing and your feedback.

I didn't change anything that could explain the issue you are seeing with Jupiter Lander. However, if the timing is very close to the limit (which it seems) then the temperature of the C64 could effect the behavior. E.g. there may be visible artifacts on the screen when you just have powered on the C64, but these may be less visible when the C64 has been running for a while. 
So when comparing you could try to switch between the two firmware releases a few times to see if you get a consistent behavior.

KimJorgensen, 2022-03-30:
A new firmware has been released to address the issue with Jupiter Lander. Since there is already an existing issue for it (#94), I will close this one.
Please feel free to create a new issue if you detect any other problems

closed 2022-03-30 by KimJorgensen