KimJorgensen/KungFuFlash #120: D64s rarely work, but CRTs always seem to be fine

ogg1e, 2022-04-04:
Does it make sense that D64s rarely work for me, but a CRT of the same game always does?  Am I doing something wrong when loading a d64?  Are these just bad images?

 The games load to various points (crack intro, menu selection etc) but never make it to the actual gameplay.

I've tried it with 2 different KFF carts and 2 different machines C64 NTSC and C64C Pal.

I know that NTSC support isn't as good as PAL.  But they behave the same on either.

For example, this one works fine:
https://csdb.dk/release/?id=100018

But these fail on either machine and both carts:
https://csdb.dk/release/?id=70478
https://csdb.dk/release/?id=162122

Can someone else verify if these D64s work/fail for them?

Sebby75, 2022-04-04:
from the main page
"Disk drive emulation is using kernal vectors and will not work with fast loaders or software that uses direct hardware access which a lot of games does."

Tested both linked disk images and when emulated drive number is set to 9.. both produce garbage on the screen.. when emulated drive set to 8.. both start fine, you get start screen and menus... but seem to get stuck on gray screen on mission start.

ogg1e, 2022-04-04:
But if the game uses direct hardware access, then why would the CRT game I posted work fine?

Sebby75, 2022-04-04:
CRT is a cartridge version of the game, it doesn't need to access floppy drive at all...   
"hardware access" in that sentence refers to physical floppy drive hardware which, game is trying to talk to via IEC port, and there is nothing connected to it..

ogg1e, 2022-04-04:
Interesting. I didn't realise that there would be so many issues with D64. I've seen this on other games as well.

closed 2022-04-05 by ogg1e