KimJorgensen/KungFuFlash #122: CD32F405RGT6 does not start.

jgilcas, 2022-04-13:
In theory it is an exact clone and can be programmed, once programmed the led starts flashing fast but it is exactly the same when I connect it to the C64.
I just wanted to know if anyone has tried it, I have assembled two and with the same result, even changed the transistor for a 945 and nothing has changed.

Thanks.

KimJorgensen, 2022-04-13:
The LED is flashing rapidly when no valid C64 PAL or NTSC clock is detected.
You could try to start the diagnostic tool in [USB only mode](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.27) and see the measured phi2 clock frequency via a terminal emulator program on the PC.

I have no experience with the CD32F405RGT6 so I'm not sure if it will work as a replacement

jgilcas, 2022-04-13:
Thanks for the reply, is not working at all, is not entering in debug mode so this is not a valid replacement.

Close the issue please.

Thanks again.

KimJorgensen, 2022-04-13:
OK, thanks for testing

closed 2022-04-13 by KimJorgensen