KimJorgensen/KungFuFlash #124: _DMA / DOT Clk  (question/advice)

anonymouz, 2022-04-26:
(I'm not aware of a KFF forum, so I'll ask here, please redirect me if I'm wrong.)
(also... I'm a hardware beginner...)

The _DMA and DOT_Clk lines are not connected...

- What hardware and/or design change would you recommend to connect _DMA and support REU ?
- Could the DOT clk be used for bus timing instead of PHI2 ? Would it be more accurate?
- Did you use a development board? Which one?

Thanks,
Onno

uffejakobsen, 2022-04-27:
I'm not Kim - 

As far as I remember there are not that many GPIOs left on the selected MCU model...

But you could sacrifice the LED - again as far as I remember that pin is 5v tolerant (I did not do and fact checks before writing this)

Running the current handler system on the dot clk will give a lot of challenges i think - but again Kim should know his design better than me :-)

/Uffe :-)

KimJorgensen, 2022-04-27:
There is already an issue for REU support with some comments https://github.com/KimJorgensen/KungFuFlash/issues/68

The DOT clock could be used but then it would not be possible to have the microcontroller generate the correct clock for the USB to work. There is also a phase shift between the clocks that changes when the C64 is getting warm (https://retrotinker.net/who-turned-the-clock/) which could cause problems, but this is not something that I have tested.

I did use a development board for a testing a few things but quickly moved on to my own PCB. It was just a random cheap board from AliExpress

KimJorgensen, 2022-05-13:
@anonymouz If you have further questions, feel free to post them in this thread https://www.forum64.de/index.php?thread/96247-kung-fu-flash-cartridge/ or create another issue

closed 2022-05-13 by KimJorgensen