KimJorgensen/KungFuFlash #125: Test with GD32F405RTG6 and DFU mode

repetto74, 2022-04-30:
I have mounted a GD32F405RTG6 on the Kung Fu Flash PCB. 
I would like to program it via USB but despite having shorted J1 and B0 / +3.3v of J4 I cannot get in DFU mode with the Dfuse software.
Is there any way to force DFU mode or is that GD32 not compatible with DFU mode?
I would like to explore all possibilities and trials before buying an ST link V2 programmer.
Thanks
Rick

jgilcas, 2022-05-03:
I have tried with two of them, they programmed well (I did it with the stlink v2) even the led blinked but in the C64 always black screen, I don't know if they are the two of mine that are false or that they are not totally compatible.

jgilcas, 2022-05-03:
I tried with DFU also but the drivers issue is different, you have to install the GD32 DFU drivers, the STM32 ones are not valid.

repetto74, 2022-05-03:
Hi Kim
That’s s bad news indeed 😞. I heard that this GD32 does not seem to be a valid replacement for the STM32 equivalent. 
That’s bad as I do not really know when this chip will be available again. 
I can try my luck with the STlink V2 programmer but I think it will end with a non working KFF also for me. 
Thanks for your feedback !
Rick

repetto74, 2022-05-03:
Hi Kim
Thanks I will update drivers with those suitable for the GD32 and give it a try. 
I will report if I can successfully program it and test. 
Thanks. 

Rick

repetto74, 2022-05-06:
> Using same board i replaced GD32 with a STM32, reprogramed with stlinkv2
> and are working like a charm. Unfortunately i think is not a good
> replacement :(

Yes I think that's a fact :-(. I have ordered an STM32 also and hope to
resolve it then ;-).

chuckcaley, 2022-05-10:
I to have tried the GD32 and all i got also was a blank screen and a blinking led. Programmed ok on the STlink.

repetto74, 2022-05-10:
Hi chuckcaley,
Conclusion is that GD32 is not a direct replacement for the SMT32 at least not for the KFF project.
Grab an SMT32 and it will work good ;-).

KimJorgensen, 2022-05-13:
Thank you for testing this.
I have added a note to the [release](https://github.com/KimJorgensen/KungFuFlash/releases/tag/rev.2) that the GD32F405 will not work as a replacement

closed 2022-05-13 by KimJorgensen

Sgw32, 2022-06-04:
Any chance GD32 version will exist? 
I've ordered 10 pcs cartridges not so long ago and now I'm struggling with the same trouble 😄 
I'll start debugging it this summer but I'm not sure I'll do it until the end. Here is an errata for GD32F405:

http://easyelectronics.ru/img/ARM_kurs/STvsGD/Migration/compatibility%20sumup%20between%20GD32%20and%20STM32_V2.0.pdf

repetto74, 2022-06-04:
Just to confirm the incompatibility again. I have replaced the GD32 with the STM32 and now the KFF will just work flawlessly on first boot :-)

Sgw32, 2022-06-04:
Yes, but STM32 chips are now expensive and hard to find.

repetto74, 2022-06-04:
I sourced it from Aliexpress for 12 euro which is still acceptable if you are not producing large quantities.

Sgw32, 2022-06-04:
It's good, but often they supply defective parts along with working - and that's what I really beware. Especially AVR series, but STM32 too. Anyway LCSC is still better but more expensive. 
And it allows automated assembly.

repetto74, 2022-06-04:
Here is the Aliexpress shop where I sourced the STM32.
https://www.aliexpress.com/item/1005003616330034.html?spm=a2g0o.order_list.0.0.7ad01802IGFixS
Comments are positive and it looks they deliver genuine STM components.

Sgw32, 2022-06-04:
Thanks! 
I have ~2-3 STM32s in my stock, however would be great to adopt GD32... I've started debugging and seems like it is stuck here:

if (!(TIM1->SR & TIM_SR_CC1IF) || TIM1->CCR1 < 160 || TIM1->CCR1 > 170)

CCR1 is 0 on GD32, TIM1 SR has bit CC1IF reset and that's why it does not work. 

GD32 has TIM0 instead of TIM1, however *real* addresses in hardware are the same, same for  registers. 
For now I do not know why it does not work. If someone can adopt GD32 to this project, I can pay for this work 
@KimJorgensen do you have any ideas, maybe you can fix firmware for GD32 for some price?

KimJorgensen, 2022-06-06:
@Sgw32 No, I have never tried to use a GD32 and it is not really something that has my interest. However, if somebody manages to get it working I'll be happy to add a link to their port.

Sgw32, 2022-06-28:
After some work my colleague has found that ARR register is not set for GD32 and that's why it does not count 1 MHz clock. That was the first issue it just blinks. Now it looks like it boots and even reset is not zero, but there is still a black screen and GAME=0, EXROM=1. I hope we'll make a fix for GD32 soon, after that I'll publish PR.

Sgw32, 2022-07-07:
GD32 version is almost ready. Working +- correctly on C64(not C64C, there are some troubles on its motherboard). I'll make pull request within a week. 
For now I've seen that at least EasyFlash cartridge is not functioning on it.

KimJorgensen, 2022-07-07:
@Sgw32 Just to avoid misunderstandings; I'm not interested in maintaining new variants of the Kung Fu Flash hardware and therefore will not accept such a pull request.
However, you are of course free to do your own fork, and as said, I'll be happy to link to that once you get it working

Sgw32, 2022-07-10:
@KimJorgensen @repetto74 here is GD32 version with a release file: 
https://github.com/Sgw32/KungFuFlashGD32

You can mention it on a page if needed. Hardware is the same.

repetto74, 2022-07-10:
Hi Sgw32 :-)

That awesome!! :-D. I will mount another KFF with the GD32 I have in stock and give it a try. I can be the Beta tester and report how it works.

repetto74, 2022-07-10:
Is there already a patched firmware available or do we need to compile it?

Sgw32, 2022-07-10:
Check releases

KimJorgensen, 2022-07-10:
@Sgw32 Great. I have added a link to your fork here https://github.com/KimJorgensen/KungFuFlash/releases/tag/rev.2

repetto74, 2022-07-10:
Got it! Thanks Sgw32 :-D!