KimJorgensen/KungFuFlash #126: KFF (v1.40) not detected on Windows

beatmaster13, 2022-05-29:
Hi, Kim! I try to use EF3 USB Utilities, but Win7 does not detect KFF as COM port

[url=https://ibb.co/9vY6k29][img]https://i.ibb.co/9vY6k29/ERRORUSB3.png[/img][/url]

KimJorgensen, 2022-05-29:
I did not have this problem, but according to Metallic on [forum64](https://www.forum64.de/index.php?thread/96247-kung-fu-flash-cartridge/&postID=1549026#post1549026) you can use [Zadig](https://zadig.akeo.ie/) to install the "USB Serial (CDC)" driver and Kung Fu Flash should show up as a COM port.

beatmaster13, 2022-05-30:
Metallic's method worked!
Just to mention... when I started the server on the KFF, it would disconnect from the PC. I fixed it by reverting the 1.26 update.

Thanks

KimJorgensen, 2022-05-30:
Great that it worked :)
I not sure what you mean with "started the server on the KFF". If you start a program on the KFF that requires EF3 USB support you should select "Load" from the file options menu. The default "Mount and load" will not work.

closed 2022-06-26 by KimJorgensen