KimJorgensen/KungFuFlash #128: Build Firmware: Error-> collect2: error: ld returned 1 exit status

soenkesmiletzki, 2022-06-03:
I want to rebuild the firmware, but i get an error (Ubuntu 20.x):

make
arm-none-eabi-gcc build/main.o build/ff.o build/ffunicode.o build/startup_stm32f405xx.o -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -specs=nano.specs  -lc -lm -lnosys -Wl,--gc-sections -Tstm32f4xx/STM32F405RGTx_FLASH.ld -Wl,-Map=build/KungFuFlash_v1.XX.map,--cref -o build/KungFuFlash_v1.XX.elf
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: build/KungFuFlash_v1.XX.elf section `.rodata' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 1280 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:143: build/KungFuFlash_v1.XX.elf] Error 1

Any idea to fix this? Missing some libs?

KimJorgensen, 2022-06-03:
I guess that you are using a different gcc compiler that produces a larger binary. You can try with the version I'm using: [GNU Arm Embedded Toolchain 10.3-2021.10](https://developer.arm.com/downloads/-/gnu-rm)

Sgw32, 2022-06-05:
> I guess that you are using a different gcc compiler that produces a larger binary. You can try with the version I'm using: [GNU Arm Embedded Toolchain 10.3-2021.10](https://developer.arm.com/downloads/-/gnu-rm)

I can confirm that it is working on 10.3. Also I've used latest CC65 and it issues "no COLOR_VIOLET" problem. I've made a workaround if needed.

closed 2022-06-26 by KimJorgensen