KimJorgensen/KungFuFlash #129: Additional device IDs outside of jumperable range

Gabeki17, 2022-06-25:
Would it be possible to add device IDs above 11?

Like it would be very useful to be able to assign an ID no other device is using for sure.

Alos if the KFF drive could be always available or could be switched of.

Thx!

All the best,
Gábor

KimJorgensen, 2022-06-26:
It would be possible but I'm not sure why that would be useful. Could you describe what you are trying to do that doesn't work with the current solution? 
If you don't want any drive emulation you can disable Kung Fu Flash (F8) or select a cartridge. You can also load a PRG file without drive emulation by selecting "Load" in the file options menu (SHIFT+RETURN)

Gabeki17, 2022-06-26:
> It would be possible but I'm not sure why that would be useful. Could you describe what you are trying to do that doesn't work with the current solution? If you don't want any drive emulation you [...]

Hi,

Sure,

So I have 3 real drives attached.
Aloso an sd2iec.

This means if i would like to use the KFF as a drive I can not.

Of course it is possible, even for real drives to assign an ID above 11, but only after switched on and done manually or by a program. But! KFF can not be pluged in after, while the c64 is on already.

So If KFF could listen above 11, would mean some extra compatibility and extended usage possibility.

KimJorgensen, 2022-07-04:
From [version 1.41](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.41) it is possible to set the device number from 8 to 15.

/Kim

closed 2022-07-04 by KimJorgensen