KimJorgensen/KungFuFlash #132: EF3USB (KungFu Flash) broken by later fw

InsertDiskTwo, 2022-08-31:
Since getting my KungFu flash one of the functions I like to use is connecting it to PC via USB using a special certain of EF3USB to write .d64 images to a real 1541 connected to the C64.

Today I upgraded the fw to the latest version, and since I am not able to connect the PC and C64 anymore. I "solved" this by flashing the KungFu with an old version. Perhaps someone can look into it. Thanks.

KimJorgensen, 2022-08-31:
What exactly are you trying that doesn't work with the latest firmware and what is the error message?
Have you started the Kung Fu Flash menu and got an error when sending the program from the PC with the EF3USB s command? Or are you trying something else?

KimJorgensen, 2022-09-01:
So is `ef3usb comX 0` the first thing you do on the PC side? You are not sending ef3usb.prg to the C64 with the `ef3usb comX s` command?
What are you doing on the C64 side? Did you just start the launcher by pressing the menu button, or did you do something else?

InsertDiskTwo, 2022-09-01:
Hi,
I launch EF3USB.prg directly on the C64 from the SD card in the Kung Fu Flash. I use the EF3USB com3 0 command to make sure communication is good. Works fine with FW 1.32 but not with latest. 
Am I doing it wrong?
Thanks!

KimJorgensen, 2022-09-01:
There is no need put the ef3usb.prg on the SD card as it can be send to the C64 with the s command. However, if you want to run a program that requires USB support with FW 1.36 or newer, you need to use the "Load" option (see https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.36)

InsertDiskTwo, 2022-09-01:
I confirm it works. I now send EF3USB directly from the PC as well. Did not know that was a possibility. Thank you for your help, much obliged!

KimJorgensen, 2022-09-01:
Great that it works. I'll close this issue but feel free to open another if you detect any other problems

closed 2022-09-01 by KimJorgensen