KimJorgensen/KungFuFlash #135: 128k Super Snapshot ROM support

jpcompton, 2022-09-16:
Recently-released [Super Snapshot V5 v5.31](https://github.com/adrianglz64/snappyrom/releases/tag/v5.31) aka "SnappyROM" fails in a manner the developer says is consistent with the full 128k ROM not being loaded properly.

I am on KFF 1.41 on NTSC C64.

Steps to reproduce:

1. Load appropriate .crt to KFF (snappyrom-5.31-ntsc.crt in my case)
2. C64 reboots to Snapshot menu
3. Press F5 to enter Turbo Macro Pro
4. See weird ugly garble and/or crashing

Other basic Snapshot features (those lower in the ROM?) seem to continue working fine.

jpcompton, 2022-09-16:
https://user-images.githubusercontent.com/51372049/190692813-2868be63-9be6-4523-8124-ea7b416bde71.MOV

gillham, 2022-09-16:
Check pull request #136 for fix that works for me.

gillham, 2022-09-16:
[KungFuFlash_v1.XX.zip](https://github.com/KimJorgensen/KungFuFlash/files/9586430/KungFuFlash_v1.XX.zip)

If this file attachment works, here is a binary .upd file (zipped up) if you have trouble building.

KimJorgensen, 2022-09-16:
Support for 128k Super Snapshot ROM has been added to the [latest release](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.42)

jpcompton, 2022-09-16:
Thank you. TMP now starts but appears unstable, I'm trying to tag in Adrian and some other people who both have KFF and TMP knowledge to provide more useful feedback!

KimJorgensen, 2022-09-16:
OK, unstable how? (how to reproduce)
Did you also try with @gillham's PR to see if that makes any difference?

jpcompton, 2022-09-16:
The buffer sometimes starts with garbage, sometimes starts clean.

In either case, it will crash fairly quickly just upon cursoring around the editor, as seen in this vid:

https://user-images.githubusercontent.com/51372049/190713778-417fb8e2-12ad-4117-a78d-909cfa51413a.mov

KimJorgensen, 2022-09-16:
KFF does not clear the C64 or the SS5 memory. Could that be cause an issue?
I have tested this on both a PAL and NTSC C64 and did not get such a crash

jpcompton, 2022-09-16:
Here's a vid of a startup with a corrupt buffer: scrolling to the bottom triggers the crash.

Are you able to enter a good deal of text without a crash on your side?

https://user-images.githubusercontent.com/51372049/190714827-9a905058-82f7-4f36-a44f-94bfd282d679.MOV

KimJorgensen, 2022-09-16:
Yes, I can enter several "pages" of text and scroll without any crash.
[image]

Maybe this is a timing issue. Do you have access to another C64 you could try it on?

jpcompton, 2022-09-16:
I think you're on to something, although not the easiest to debug, as the closest alternate machine (a 64C) behaves _worse_ and I can't even get the KFF menu screen. Will dig for a third option.

jpcompton, 2022-09-16:
Okay!

Of the three working straight 64s I have nearby and feel I can trust:

NTSC breadbin: flaky behavior as documented in videos

NTSC 64C 1: can't even boot properly to KFF menu (_I can't swear to whether or not this has always been true or is only true in 1.42, if you want me to roll back some versions and test, let me know_)

NTSC 64C 2: Seems stable!

jpcompton, 2022-09-16:
FWIW for timing hints: I have a 1541 Ultimate II (not plus) which currently works on both of those 64Cs with settings

valid after PHI2: 100ns
edge recovery: enabled

It does _not_ work on the breadbin with those settings.

KimJorgensen, 2022-09-16:
There can be quite a difference in timing requirements between C64s, even with the same board revision. For that reason I usually test with eight different PAL machines when doing timing critical changes. However, I have only access to two NTSC machines so I have not fine tuned that to the same degree.

jpcompton, 2022-09-16:
Understood.

Either here or as a separate issue, would it help if I ran some tests on these three machines? Timing outputs from https://vice-emu.pokefinder.org/wiki/Testbench tools, that sort of thing?

KimJorgensen, 2022-09-16:
Thank you, but I have not any tools that can help remote debug such problems. It usually requires some experimentation to figure out what is wrong and to find a fix that would work on all machines. I could try to make some alternative firmware versions for you to test, if you are interested.

Regarding NTSC 64C 1: It could be a timing problem, but if the KFF is unstable while pressing the buttons, it could also be that the edge connector needs cleaning.

jpcompton, 2022-09-16:
I don't think it's a contact issue. The 64C 1 won't bring up either the KFF menu or a preloaded KFF disk/cartridge image. I _am_ able to reset to BASIC using the MENU+SPECIAL hold and I see the debug light flash. So if that information coming over the USB port might be useful, please let me know how to get at it (probably easiest for me to put a Linux laptop close enough to reach it with a Mini-USB cable) and I can report.

And yes I am happy to test alternate firmwares on the machines.

gillham, 2022-09-16:
I tested on a NTSC breadbin with a high serial number.  5 million+
So with 1.42 I see some instability with a KFF v1.2 & v1.3 (per TFW8b naming).
With my version I seem to be pretty stable with the KFF v1.2 but not with the KFF v1.3.
With the KFF v1.2 I see Phi2 frequency of around 1022679
With the KFF v1.3 I see Phi2 frequency of around 1022686

I will test a couple other units. (64C etc)

gillham, 2022-09-16:
My machines are all NTSC.
I only really tested Super Snapshot (freeze) on the 2nd 64C.
Tests with TMP were fairly consistent results.
Tests with SS5 freeze seemed less consistent. (again only tested on one machine)

The 64C units were stable with both cartridges and both firmware versions.
The v1.42 firmware isn't stable on either breadbin with either cartridge.
My patched firmware only seemed stable on my initial test / dev setup breadbin
with the v1.2 cartridge.  The 2nd breadbin was close but not stable.

Super Snapshot itself (pressing freeze button) wasn't exactly stable on the 2nd
64C (the last machine I tested) when I decided to pop into the monitor.

My patch:
1st 64 v1.3 Phi2: 1022686  TMP is unstable
1st 64 v1.2 Phi2: 1022679  TMP seems stable  <-- my original test/patch setup

2nd 64 v1.3 Phi2: 1022647  TMP fairly stable (dumped to basic once possibly?)
2nd 64 v1.2 Phi2: 1022644  TMP is unstable / almost stable. (mostly, first time failed)

1st 64C v1.3 Phi2: 1022646  TMP stable
1st 64C v1.2 Phi2: 1022640  TMP stable

2nd 64C v1.3 Phi2: 1022639  TMP stable, SS5 stable
2nd 64C v1.2 Phi2: 1022634  TMP stable, SS5 not stable (freeze slow or blank screen)

v1.42:
1st 64 v1.3 Phi2: 1022686  TMP is unstable (first test was stable, then crashed)
1st 64 v1.2 Phi2: 1022679  TMP is unstable (loaded with corrupt memory)

2nd 64 v1.3 Phi2: 1022646  TMP doesn't load
2nd 64 v1.2 Phi2: 1022640  TMP loads, file loads, memory corruption with paging/scrolling

1st 64C v1.3 Phi2: 1022647  TMP stable
1st 64C v1.2 Phi2: 1022641  TMP stable

2nd 64C v1.3 Phi2: 1022639  TMP stable, SS5 unstable (freeze button blank screen)
2nd 64C v1.2 Phi2: 1022633  TMP stable, SS5 stable

I'm not sure if this is telling me too much.  There seems to be ~6Hz difference between the two cartridges regardless of which machine is used. My 1st breadbin is close to 40Hz higher than the other machines, but the 2nd breadbin is unstable and it is much closer to the 64C frequencies.

adrianglz64, 2022-09-16:
Howdy!
Quick comments about the 128K ROM:

A couple of original SSv5 features are basically files that are "loaded" from ROM to RAM and then run:

- Disk copier (F3 in startup menu, then option #2)
- Basic plus (>£ in basic)

I moved both of those to the upper 64K.  TMP also lives as a file in the upper 64K.  Like the other files, it's loaded to RAM and then runs from RAM.

I have an NTSC breadbox that fails to load TMP in a manner similar to what has been described above.
The weird part is that both the disk copier and basic plus seem to work fine on it, and those two features are also retrieved from the upper 64K.  The only difference that comes to mind is that TMP is quite a bit bigger than those other two.

TMP works fine on my NTSC c64c.

I'll try writing a small test program that reads and verifies each ROM bank a few times.  Maybe that'll shed some light on the problem.

KimJorgensen, 2022-09-17:
Hi!
I have made some test firmware that that I would like you to test:
```
vTST6: Optimized SS5 code
vTST7: vTST6 + advance timing
vTST8: vTST6 + retard timing
```
[SS5_test_fw.zip](https://github.com/KimJorgensen/KungFuFlash/files/9592576/SS5_test_fw.zip)
All 3 versions works fine on my machines, and so did v1.42, so this is a bit of guesswork. But let me know the results.

@gillham Regarding the Phi2 measurement; the KFF is not a super accurate frequency counter, and the frequency may vary depending of the mounted clock crystal. The frequency of the C64 may also slightly drift as the computer gets warm.

adrianglz64, 2022-09-17:
Preliminary report:
TST6 and TST7 seem stable on my breadbox that fails to load TMP with 1.42
TST8 is _almost_ stable.  I've had it fail once while loading the disk copier and a couple of times while loading TMP.

I hope to have that test/checksum tool ready soon.

jpcompton, 2022-09-17:
**NTSC breadbin** (was: flaky)
TST6: Seems stable
TST7: Seems stable
TST8: Seems stable

**NTSC 64C 1** (was: no KFF functionality)
TST6: no KFF functionality
TST7: no KFF functionality
TST8: no KFF functionality

**NTSC 64C 2** (was:stable)
TST6: Can't use any freezer carts (typically start to black screen). Unstable with other disk/cart images, some lockups in KFF menu mode as well.
TST7: Can't use any freezer carts (black screen, other glitches).  
TST8: Seems stable

KimJorgensen, 2022-09-18:
Thank you for testing and reporting back. This is very useful, now that I cannot reproduce the problem myself.
@jpcompton Are both NTSC 64C 1 and 2 both short boards (Assy 250469)? My NTSC C64C is a long board (Assy 250466)

Here are two new firmware versions that I hope you have the time to test:
```
TST9: TST8 + retard timing for KFF menu
TS10: Timing between TST6 and TST8, also for KFF menu
```
[SS5_test2_fw.zip](https://github.com/KimJorgensen/KungFuFlash/files/9593491/SS5_test2_fw.zip)

For previous test versions I did not change the timing used by the menu, hopefully these versions will allow NTSC 64C 1 to work.

gillham, 2022-09-18:
I tested two C64 breadbins, two 64C, with two KFF.
One C64 with v1.3 KFF froze loading disk copy on vTST6.  It only happened once and I couldn't reproduce it.

Both 64C with vTST7 just gave me a black screen on selecting SS5 original or Snappy.  The KFF menus seemed fine.

Every other combination of vTST6 through vTS10 were stable in my (admittedly quick) testing. Though this same amount of testing found issues with v1.42 & my patch.

My procedure more or less:
Flash KFF test code
Select snappy rom
F1 / yes clear memory
F3 / disk copy / exit (3 times)
F5 TMP
<-@cd;disk.d64
<-L testfile (~60 lines of setting NMI/restore to TMP, messing with colors on some raster lines)
<-3 S to launch it / restore key to return to TMP (repeating 2-3 times)
<-3 S then freeze button to SS5 menu and then return from it
add lines to top and bottom of file. (<-2 filler mostly)
page around with function keys
re-assemble / run a few more times
reset button on KFF
load / exit disk copy 3 more times
F5 for TMP
etc.

Also was switching between a couple test files in TMP.  Basically I tried to do a similar set of operations in the same order each time to try for consistency.

Also re-tested v1.42 on a C64 breadbin that was having issues consistently and it was still having issues.  Even with clearing memory, just loading TMP had garbage in the buffer already.

The vTST6 may be close to the edge on some timing as I did have one issue, but wasn't reproducible.

The vTST7 clearly is broken in the 64C units. I'm surprised my breadbin that had trouble (once) with vTST6 worked.  So my vTST6 issue is maybe bogus since I couldn't reproduce.

The vTST8/vTST9/vTS10 were all stable. (and vTST6 was close)  Does that say the SS5 optimization code is helping my systems more than the timing?  I tested each for about 5 minutes.  Later tonight I can probably test one or two combinations longer / leave the machine powered on longer etc.

gillham, 2022-09-18:
Really since most of my testing is just to say "did code get transferred from ROM to RAM correctly" the test/checksum tool Adrian is working on sounds ideal for repeatedly testing transfers.

KimJorgensen, 2022-09-18:
@gillham Thank you for the description of your test procedure

I'll try to explain what is going on with the optimizing and timing adjustments:
From the tests and my experience with PAL machines, it seems that the control bus (ROML, IO1, etc) is stable later on a 64C than a breadbin. The breadbin, on the other hand, requires data to be written earlier to the data bus

By optimizing the code to make it execute faster will allow the data to be output earlier and makes it stable on the breadbin. Adjusting the timing to read the control bus later will make it stable on the 64C. However, reading the control bus later leaves the code less time to output the data

jpcompton, 2022-09-18:
(Am adding more details to description in part to help me keep this stuff straight!)

NTSC 64C 1, has 6526B and CMD Jiffy (was: no KFF functionality)

TST9: Started out seeming stable and I was able to explore Snapshot 5.31 features correctly, but then experienced crashes intermittently when in KFF menu, Snapshot freeze mode, etc. Then booted to black. Heat? Speed drift? No idea.

TST10: Initially, booted to BASIC screen with 30719 bytes free. Menu button crashed. I rejected failure and kept prodding, and by removing the SD card I was able to I can coax it to the "please insert SD card" KFF menu. After that it acts like one of the "regular KFF stuff works but no freezer cart works" scenarios. Freezer cart images either boot to black or 30719 bytes, but other tasks seem stable.

NTSC 64C 2, has 6526A and stock ROM (was:stable)

TST9: seems stable
TST10: seems stable

KimJorgensen, 2022-09-19:
@jpcompton Thank you for testing and sharing your results. Did you by any chance also test TST9 and TST10 with your breadbin?
I'm not sure that I can make a release that works with your NTSC 64C 1 without further optimization, which is quite tricky by now. However, I made a firmware version just to see if adjusting the timing further will make it stable. This version is not stable on my breadbin
```
TS11: TST9 + retard timing further
```
[SS5_test3_fw.zip](https://github.com/KimJorgensen/KungFuFlash/files/9601340/SS5_test3_fw.zip)

Regardless of the result I think I'll release a version based on TST9 or TST10 which seems to work on more machines than v1.42. I have ordered parts for an NTSC 64C shortboard so hopefully I can reproduce the problems myself soon

jpcompton, 2022-09-19:
I had not run on the breadbin but have now:

**NTSC breadbin**

TST9: 
TST10: seems stable

I'm a little leery of running something that might break on the breadbin since that's been my reliable rescue machine and my cart did not come with the USB-enabling jumpers installed. What is TST11 _likely_ to work on? And/or is there a new secret way of emergency-flashing firmware if I need to?

KimJorgensen, 2022-09-19:
Thank you and I fully understand. Unfortunately I can't guarantee that you won't "brick" your cartridge.

There is no secret way to flash the firmware other than what is described here https://github.com/KimJorgensen/KungFuFlash/blob/master/firmware/README.md

jpcompton, 2022-09-19:
Bit the bullet, I hope this proves useful:

**NTSC breadbin**

TST11: Unstable, @ symbols randomly creep in to the TFT interface ("Progr@mming flash memory.", for example), Snapshot 5.31 initially loaded okay and first attempt at an upper-bank function (entering >£ in BASIC for more BASIC extensions) went fine, but subsequent attempts to enter the disk copier and TMP failed. Freezer mode seemed okay, though. Other functions seem fine, was able to switch between multiple 300k+ EF CRTs without obvious issue.

**NTSC 64C 1**

TST11: >>>!!!seems stable!!!<<<

**NTSC 64C 2**

TST11: also >>>!!!seems stable!!!<<<

KimJorgensen, 2022-09-20:
Thank you. It confirms that it is indeed the timing that needs adjusting to support your stubborn 64C, and that will unfortunately not work on breadbins.

I have created a [new release](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.43) based on TST10, and will try to come up with a solution that will work on all C64 models

jpcompton, 2022-09-27:
Probably no surprise to you, but 1.43 is unstable on NTSC 64C 1 and stable on NTSC 64C 2.

adrianglz64, 2022-09-28:
Very late to the party, but here's a test tool that computes a checksum for the ROM and then reads and compares each 16K bank several times:
[ssv5romtool.zip](https://github.com/KimJorgensen/KungFuFlash/files/9662690/ssv5romtool.zip)

1.43 is stable on both my breadbox and my c64c:

[image]

I tried the TST8 firmware which was a little unstable on my breadbox and I got some errors on banks 4-6.  Oddly enough banks 0-3 and 7 were fine.  I'll be releasing the test tool as soon as I write a readme for it.

gillham, 2022-09-28:
Sweet!  I'm going to try that out today.  Thanks for writing this tool.

KimJorgensen, 2022-09-29:
@jpcompton Yes, no surprise but thanks for verifying the release.

@adrianglz64 Great, that is very useful when testing timing adjustments.
KFF stores the first 64kB (banks 0-3) of a CRT in ram which is faster to access than the other banks which is stored in flash. I'm not sure why bank 7 would be different from banks 4-6, except that most of the contents are $ff

gillham, 2022-09-30:
Well I found one of my breadbins that occasionally gets errors with @adrianglz64’s testing tool. Also one of my 64Cs seems like it doesn’t always power up correctly with the KFF. The menu still pops up but launching the cartridge is a black/dark screen.  Not just SSv5 though. Powering off and on once or twice typically boots it.  I don’t see that with EF3 or other cartridges. 

Would it be feasible to add some ability to mess with timing to the advanced menu?  And allow holding a button on power up to load failsafe / default timings? (Or just adjust timings in ram and never write / save so a power cycle clears)

Maybe holding a button on boot could select alternate timing also?

jpcompton, 2022-09-30:
@gillham Maybe you should post a screenshot of the errors so Kim and Adrian can compare notes on the affected bank(s)?

And yes, I can confirm that I haven't seen issues with EF3 or with Ultimate 1541 II/II+ on the 64Cs which are/have been unhappy about KFF.

gillham, 2022-09-30:
I’ll run it some more this weekend. The first failures I got were in banks 4/5/6. Just a couple each.  I’ll test more.

jpcompton, 2022-10-07:
(crazy side note: you know the breadbin which has been rock-solid with KFF throughout except in the extreme test builds? Turns out that guy doesn't work with either of the EF3 carts I own!)

KimJorgensen, 2022-10-24:
A [new release](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.44) has been created with some performance optimizations and a tool for adjusting timing, as suggested by @gillham

gillham, 2022-10-25:
Thanks @KimJorgensen for adding this feature!  I tested it in my C64C that typically showed a black screen on initial power on with a previously selected cartridge, but showed the menu with the button.  It would typically be ok after a power cycle or two.  Well with v1.44 and just bumping it to +1 it powers up fine. I left it off for 4 plus hours after changing that setting and it launched Super Snapshot straight from power on.

Without the offset even Epyx Fast Load would show a black screen. (So not just a Super Snapshot issue).  I’m kind of surprised just +1 is enough but it was fairly close since it would work with some power cycling.

I will test with a couple other units. Including the bread bin I occasionally got a few errors with using @adrianglz64 rom tool.

gillham, 2022-10-25:
I tested this on a couple C64C and a couple of breadbins. And maybe one short board in a breadbin case temporarily.  The “0” setting was ok for one breadbin but got a couple of errors with the rom tool on another.  The “-1” was clean on both. One C64C needed “+1” to be error free and post. The other seemed ok on “0”.  Both worked fine with “+1”

So should we find the highest number and lowest number that is clean and set it in the middle?  For example one breadbin was good from 0 to -8 and the other was good from -1 to -9. Should I use -5 since that is more or less in the middle of the clean/ good range?  Or should we use the closest clean value to 0?

Similarly the 2 x C64C & short board (temp mounted in a breadbin) varied a little bit had a narrower range. Like +1 to +3 was clean. So would +2 be better than +1 as it is more in the middle?

 Anyway this adjustment seems really nice for having a solid timing number for each box.  Of course moving a KFF between my breadbin & C64C means reconfiguring the timing offset but it is super easy to do.  And now my C64Cs both can power on reliably with the KFF instead of a black screen and having to power cycle a couple times.

KimJorgensen, 2022-10-25:
@gillham Thank you for your detailed test report, it is really helpful when I only have a limited set of machines for testing.

So far I've tried to pick an offset that would work on most machines, which is admittedly difficult with only a random selection of machines available. If possible, I have also set the timing values 1 or 2 from the "edge" to allow for small variations due to temperature changes or other external factors. BTW the unit of the phi2 offset is MCU clock cycles, so 1 equals 6 nanoseconds.

Ideally, this tool shouldn't be needed, but would require additional code optimizations to have MCU cycles to spare, and of course knowing the timing requirements that would work for all machines used with the cartridge

jpcompton, 2022-10-26:
Pleased to report that 1.44 seems stable on all three of the previous test machines!

KimJorgensen, 2022-10-26:
@jpcompton Great, thanks for testing

jpcompton, 2022-10-27:
...dare we close this issue?!

KimJorgensen, 2022-10-27:
Yes, let's close this one and if there are other problems, just create a new issue.

SnappyROM seems like a really useful cartridge, so I'm glad we got it working. Thanks for all the help in solving this

closed 2022-10-27 by KimJorgensen