KimJorgensen/KungFuFlash #137: [Support] Sidplay64 - access sid files

shelterx, 2022-09-20:
First, excellent project and thanks for adding/fixing Sidplay64 support.
I don't have the kung fu flash cart yet but I will most likely buy one soon and I don't know where to ask this so pardon me if I'm posting in the wrong place.

I know I can probably put sidplay64 on a d64 image and put sid files on that image. But that limits the amount of sids i can have on each disk. naturally.
Is there any way to load sidplay64 with kung fu flash cart somehow and have it access a larger collection of sids?

pmandes, 2022-09-20:
just use sidplay64-sd2iec version and browse the entire HVSC content on your SD card :)

[image]

[image]

shelterx, 2022-09-20:
Very cool,  you for  your reply!

closed 2022-09-29 by shelterx