KimJorgensen/KungFuFlash #139: Demos on d64 images

shelterx, 2022-09-29:
Perhaps it's a d64 direct hw/fast- loading issue but I tried some demos from csdb and none of them worked properly.
Comaland, for example doesn't load at all, it just hangs on the basic screen.

Is there any way to workaround this?

Everything else seems fine tho', loading prg files, flashing crt images and so on.

jpcompton, 2022-10-07:
Very likely a loader issue, yes. As the README warns us:

`Disk drive emulation is using kernal vectors and will not work with fast loaders or software that uses direct hardware access which a lot of games does. Currently REL files are not supported and only a subset of the Commodore DOS commands are supported.`

shelterx, 2022-10-08:
Yes and I've searched quite a lot on how to convert d64 images into some other format but with the loaders they use it's not that easy. That's why I asked for some kind of workaround, other than buying a pi1541 of course.

jpcompton, 2022-10-08:
It's not trivial work.

But the smallish number of demos which offer sd2iec or IDE64 fixing may work. For example, this fixed release of Attitude 8 works fine from my KFF. https://csdb.dk/release/?id=223968

KimJorgensen, 2022-10-08:
Sorry, but there is no workaround. With the current hardware it is not possible to support fast loaders or software that uses direct hardware access. For that you will need a physical connection to the serial port.

The best you can do is try an alternative version of the software or use a cartridge release if available.

I'll close this as a duplicate of #50

closed 2022-10-08 by KimJorgensen