KimJorgensen/KungFuFlash #141: Freeze on scrolling

pazuzu72, 2022-10-06:
I don't know if it can be useful. I discovered a bug on the latest firmware versions (it could be the firmware or a flaw in the demo). "Wolfling Reloaded Teaser" you can find on CSDB it crashes more or less always the same, with version 1.43 and 1.41 I have tried a couple of old and it seems to be fine. I discovered it by accident. Tested on 2 C64s, one breadbin and one C64C.

shelterx, 2022-10-27:
I don't see why KFF would interfere with a loaded prg file but I have 1.41 and it freezes for me too, I think it freezes when it's supposed to change screen?
@KimJorgensen Any ideas?

KimJorgensen, 2022-10-27:
I haven't debugged the problem, but am guessing that KFF's disk emulation is somehow interfering with the demo.
When using the "Load" option from the files options menu (which disables KFF disk emulation) the demo works fine

shelterx, 2022-10-27:
Oh, nice find. I did not know the "Load" option disables the disk emulation, is that documented somewhere?
I can verify that using "Load" from the KFF menu works.

KimJorgensen, 2022-10-27:
It is documented in the [release note for v1.36](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.36)

pazuzu72, 2022-10-29:
Perfect, it works! A thousand thanks.

closed 2022-10-29 by pazuzu72