KimJorgensen/KungFuFlash #142: Load flashed CRT?

shelterx, 2022-10-08:
Maybe I've missed something but how do I load the latest flashed CRT instead of running latest loaded program or d64 image?
Let's say I flash KFF with dead test, then I load a prg file, if I hit reset then it loads the prg file again automatically, how do reboot to dead test without reflashing it?

KimJorgensen, 2022-10-08:
Kung Fu Flash does not distinguish between the type of software you launch and only remembers your last choice

shelterx, 2022-10-08:
I see. I don't know how KFF works from the technical side so pardon me if I am way wrong here. But isn't the flashed CRT still in the flash memory even if you load something else after the flash? Shouldn't it be possible to make a "Load latest flashed CRT from flash memory" feature?

KimJorgensen, 2022-10-08:
Due to the limited amount of flash (1MB), the first 64kB of a CRT is stored in the KungFuFlash.dat file. So if it is a small CRT, like the dead test, it will not be stored in flash memory.
Would it be possible to implement such feature? Sure, but it was a deliberate choice to have a simple interface

closed 2022-11-09 by KimJorgensen