KimJorgensen/KungFuFlash #143: Musik Run/Stop demo by Mahoney does not work on KFF

0x5066, 2022-10-17:
Hello!
Today I've received the KFF and it's been extremely fun to run all sorts of software i could possibly throw at it, unfortunately one of them is a demo by Mahoney, and it seems to think that it's running in an emulator...

[image]

https://csdb.dk/release/download.php?id=160166
Pressing any key as the demo suggests doesn't really do anything but stop the background from scrolling, RESTORE seemingly crashes it.

Is there anything I can do that'd make it work?

shelterx, 2022-10-20:
It's probably using a fastloader or direct hardware access.
You need a pi1541 or similar device for most d64 images to work properly. 

Or you could use sidplay sd2iec version and download the SIDs that are in the demo if you're only interested in the music. Sidplay works for most SID files, sometimes it sounds like crap tho' and sometimes you need to restart playing the SID because it stops playing.

0x5066, 2022-10-20:
Ah, that's unfortunate...
I'll see about getting an sd2iec, as I'd like to be able to play the Demo itself, since it does contain some pretty cool graphics.

Thanks for the help!

closed 2022-10-20 by 0x5066