KimJorgensen/KungFuFlash #144: KFF doesn't find microSD card

luke-70it, 2022-10-20:
Hi, I formatted a 16gb microSD card using FAT32, 4096 bytes.  For now I copied into it only firmware file (KungFuFlash_v1.43.upd) and I inserted it in KKF sd slot, but it can't detect my sd card.
Is 16gb sd too big? Do I need to change format parameters or is my KKF not working correctly?

[image]

uffejakobsen, 2022-10-20:
What model/brand microSD card have you tried ? 
I had  problems with some no-name microSD cards from China (that worked well for other uses)

luke-70it, 2022-10-20:
I tried with a 32gb HP card and with a 16gb Verbatim one.
Thank you for your support!!

KimJorgensen, 2022-10-21:
This could very well be a hardware issue, but I think you should resolve this with the seller.
I have no problems with people selling Kung Fu Flash cartridges but they should also be willing to handle the support if something doesn't work

gillham, 2022-10-28:
I would try a Sandisk or Samsung.  The Future Was 8-Bit sells the KFF with KIOXIA 16GB cards which seem to work great.

Just to confirm, you are pushing the card down until it clicks and is mostly flush with the top of the case? (if your unit has a case)  I guess it might depend on what kind of Micro SD socket that unit has, don't force it.

closed 2022-11-14 by luke-70it