KimJorgensen/KungFuFlash #147: Heavily modified C64 unstable in NTSC mode with 1.43 but works on 1.42, PAL mode seems to work though.

insanity213, 2022-11-08:
Greetings Dr. J,

I saw in your release notes that you requested feedback if we ran into issues with the NTSC/1.43 firmware combination.  I have a machine running on a 6502 with the Monotech CPU adapter board, a VIC-II Kawari and a GAL PLA (amongst some other ridiculous hacks).  I had to do some tricks with the PHI2 clock to get the KFF cart to work at all with the 6502, but I also had to roll back to 1.42 to make things stable.  I know this is an extreme corner case, but I just wanted to let you know that 1.42 was more stable in this particular Frankenstein machine in NTSC mode.  

Below is a link to a somewhat lengthy video on this machine, the last 5 minutes or so may be of interest to you when I am testing the KFF cart on 1.42 and 1.43 in PAL and NTSC modes.  The rest of the video I'm screwing with the PHI2 clock to fix the low amplitude (The 6502 PHI2 clock output seems weaker than a real 6510).  

Thanks for putting together this awesome project, you're a frickin legend!

https://youtu.be/wWZ0sSmxfMY

KimJorgensen, 2022-11-08:
@insanity213 Thank you for the feedback and the nice words.

In 1.43 I made some timing changes to try to make it stable on more NTSC machines, but it only helped for some and made it worse for others. So in the [1.44 release](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.44) I did some further optimizations and made it possible to manually adjust the timing in the diagnostic tool ([phi2 offset](https://github.com/KimJorgensen/KungFuFlash/tree/master/firmware#diagnostic)) to hopefully make it stable on all machines. 
If you have the time you could try that version. However, I wouldn't expect this phi2 offset adjustment to affect how well the clock is detected

insanity213, 2022-11-11:
Oh cool, I hadn't noticed 1.44 had dropped.  I'll play around with it.  Thanks!

closed 2022-11-11 by insanity213