KimJorgensen/KungFuFlash #148: KFF v1.43 with C-128

Andras-Simai, 2022-11-09:
Hello Kim!

So, I had flashed my KFF with the v1.43 and I found a problem.

I would have liked to play with the Prince Of Persia - it does not matter with wich versions -, the .crt could not run in C-128. Yes the C128 turned on C-64 mode, but the program cannot run. For instance: https://csdb.dk/release/?id=102540 (Mr.SID's version.)

I checked it on WinVice X128, the crt works, it can be played.

I can't see the main screen either, instead the image is blurred and freezes the C-128.

But, if I try to play with it on my C-64, it can be played.

Would you be so kind as to see it on C-128, please?

Andras-Simai, 2022-11-10:
Hello Kim!

Thank you for your help, it works!

closed 2022-11-10 by Andras-Simai

tomnic79, 2024-03-14:
Thanks a lot!!! It works beautifully!