KimJorgensen/KungFuFlash #149: Feature request: USB port for MIDI

hippi-viking, 2022-11-12:
I was wondering if it would be possible to use the USB port for MIDI in-output while outside of the firmware menu. It would make a lot of sense for ex. Prophet64 and would replicate the function of the Kerberos cartridge quite well.
(I don't know how helpful it might be but there is an [ARM-based cart](https://github.com/anarkiwi/vessel) for this, though it is wired differently than what is considered to be [*the standard*](https://codebase64.org/doku.php?id=base:c64_midi_interfaces).)

KimJorgensen, 2022-11-16:
It sounds like something that could be done, but I don't have much experience in this area.

Which MIDI interfaces does Prophet64 support? I can only find a reference of an interface that is connected to the user port, which does not correspond to what your refer to as the standard.

On the PC side, should the USB port show up as a different device (something other than a serial port)? And which software would it make sense to use here?

mwedmark, 2022-12-05:
With a software bridge as for example this: https://projectgus.github.io/hairless-midiserial/
I think the last idea seem promising, thus just open the serial port for this and use it both ways. This would be a really intereseting project to implement. It does not include any new hardware and also not new drivers for USB I guess. The hairless also seem to handle latency issues and buffer up/out to support different baud rates which in that case make it even simpler, just use anything faster f.e. 115200. Only downside that this solution always include a computer in between (C64 and synths/racks), but I guess that could be a small PI-Zero?

hippi-viking, 2022-12-06:
@mwedmark It sounds like a good compromise / first step. I don't know (yet) how difficult it is to implement USB host mode on the STM32F405 - and how problematic it is to switch to device mode when in the firmware menu (for the KungFuFlash _and_ the connected equipment!) -, so a USB host (which could very well be a RasPI) would be needed anyway at least while host mode can be implemented on the KungFuFlash to connect MIDI equipment. The downside is of course the added complexity.

hippi-viking, 2022-12-12:
I just found [this library](https://github.com/ripxorip/stm32_usb_midi), which seems to be a great candidate. I don't know yet how to interface it with the C64 though.

hippi-viking, 2024-03-02:
Hi Alex,

Sadly I didn't have much time, I am still learning to program the STM32. But there is a lot of hope, and quite a few of ready-made libraries to make it happen! I can't promise an ETA, but it is on my backlist to make the KFF a Midi interface one day.

You would need an USB-DIN adapter if you intend to use a Midi keyboard/device with the oldschool round plugs - which adapter could also be an STM32.

alexrepp, 2024-03-02:
Hi Hippi-Viking,
thanks for the response and good to hear :)
I don't have oldschool round plugs (anymore) - just two USB Keyboards.
Seen a TeensyROM Cartridge some days ago, where it seems to be realized with that USB-MIDI ... but am totally out of money this times :/ and would really be a burner at that KFF :)
Best wishes,
Alex

hippi-viking, 2024-03-07:
Hey Alex,

That's a good find. The MIDI-part of Teensy firmware should be relatively easy to port to KFF. I'll have a look, but don't promise anything as I'm still very new to this all. The good news is that it is definitely possible to make the KFF a MIDI adapter.

SensoriumEmbedded, 2024-03-07:
Hope you'll give me credit if you do that.  ;)   
Might want to check out all the TeensyROM unique capabilities.  :)
Not putting down KFF at all, it's a great cart, just think there are complementary features.