KimJorgensen/KungFuFlash #153: Support for multi-disk images

mwedmark, 2022-11-22:
Some games are sold as multiple D64-images and in the case where these work for Kung Fu Flash it would be nice to be able to traverse through several D64 files using f.e. the Custom/Freeze button. Does this make sense ot would the computer still not see the new mount as a new disk? 

If possible, either they are mounted in alpha order or there are some means to specify order, similar/same as PI-1541.

shelterx, 2023-03-18:
It would be cool but I'm not sure how useful it will be. Unfortunately many d64 images won't load properly in KFF due to the way the C64 program is written.
I don't know if the new games that are sold have that issue tho', but again, it depends on how they access the drive.

I think d64 support in KFF should be seen as nice to have but for real 1541 emulation you should get something else.

pigravity, 2023-05-15:
Oh yes, that would be handy. Its not about any 1541 emulation. Its about  useful option to change disks. IMO its a must, sadly wont be implemented.

KimJorgensen, 2024-12-29:
I'll close this as a duplicate of #49

closed 2024-12-29 by KimJorgensen