KimJorgensen/KungFuFlash #154: Multimax Cart image show bad graphics

mwedmark, 2022-12-16:
This release:
https://csdb.dk/release/?id=210760&fbclid=IwAR1DVMMp6CKh0VKl65mG1n63pMGTbCT6NR9lAmlHjlPTChMHg1jNxmN_LYs

Ultimax Collection Crt. It is working as expected when mounted in VICE but show strange graphics on, at least, 2 different C64's.

According to comments the creator has created it especially for Easyflash/KFF and it "works" on v1.28 and higher?? Maybe not tested, that is not clear. I'll do some research and add info here about versions.

Screenshot of problem:
[image]

mwedmark, 2022-12-16:
v1.28 firmware has similar issues. Looks more of less the same.

mwedmark, 2022-12-16:
https://www.lemon64.com/forum/viewtopic.php?t=74153

Seem to already have some problems with normal EasyFlash3. Should it work with KFF?

KimJorgensen, 2022-12-18:
I guess that you need to start the CRT in VIC-II/C128 mode.
You can select that in the options menu (shift+enter or long press fire).

mwedmark, 2022-12-18:
Thanks! never seen a Cart that needed that setting. Works as a charm!

closed 2022-12-18 by mwedmark