KimJorgensen/KungFuFlash #158: Stopped working

faxm0dem, 2023-02-18:
My KFF stopped working after not touching it or my c64 for some time.

The c64 just starts normally to basic prompt.
Rightmost led is on.
Sometimes I get a black screen.

What's the best way to troubleshoot ?
Is there some debugging information available on the usb port

faxm0dem, 2023-02-19:
I just checked using `dfu-util -U` the firmware is okay and corresponds to version `1.27_PAL.upd`.
I also checked when hooking up the USB port on a linux laptop, I don't see any serial port being recognised. I checked this when the KFF was plugged into the C64 (without J1) and also when it is not (with J1).

KimJorgensen, 2023-02-19:
You could try the KFF with another C64 if you have access to one. You could also try to clean the edge connector to make sure that the connection is good.
The KFF only shows up as a USB serial port when in the launcher menu, but I assume the USB port is working correctly when you can use the dfu-util program. Is there any particular reason for using such old firmware version? Otherwise you could try to upgrade to the latest  (v1.44)

faxm0dem, 2023-02-19:
Hi,

I found the root cause : one of the resistor array line was dead. I soldered a fresh one and everything works as before yay. Thanks for your help ! I'll flash the new fw now

closed 2023-02-19 by faxm0dem