KimJorgensen/KungFuFlash #160: Stuck on "Communicating with usb"

Shadowtrance, 2023-03-15:
Yep my kung fu just decided to crap itself for no apparent reason. :(
Only thing different i did related to the cart today was install my arm2sid and listen to some tunes with it from the kung fu.
I've tried everything i can think of... no sd card, different blank sd card, reflashing firmware with dfu-util and STM32CubeProgrammer multiple times to no avail.
All i did before it happened was go back to the menu to load something else and i was greeted by that.
Any ideas?

uffejakobsen, 2023-03-15:
From your description It is unclear to me what happens
- does the computer start up with KFF installed ?
- does it present anything when going into the KFF menu - is it garbled or not ?

Maybe some pictures will help us to help you..

Shadowtrance, 2023-03-15:
It seems to be trying to boot to the menu but instead of showing the menu it just has the usual menu border and bottom text but also has "Communicating with usb" yellow text and that's it. It doesn't go any further.
[image]

uffejakobsen, 2023-03-15:
I've taken a short look at the code - for some (currently) unknown reason the code does not see the expected KFF_ID_VALUE on an IO address - and continues into a EF3 main control loop where it was expected to go into a KFF main control loop...

It is a long shot - but have you tried to remove and reinsert your KFF cartridge ? maybe cleaning the PCB contacts ?

Also as we a dealing with around 40 years old computers here - they may suddenly break without notice
Do you have any other cartridges that you could test in the same computer - it could be a contact in your cartridge port that does not connect anymore...

Shadowtrance, 2023-03-15:
Yeah i tried the obvious things, taking it out, cleaning etc etc.
And being 40 would be a good point....but the only 40 year old parts in this system now are the cpu, cia's, basic and char roms and it's a rebuild to replace my "real" old one that was beyond repair.

And yeah other carts seem fine. I tried an original jupiter lander, some versa64 carts, sidekick64. No problem there.
This is really weird. Just seems to be stuck in that state and doesn't want to come out of it. :(

Shadowtrance, 2023-03-15:
Yeah i believe i do. arm2sid connects to IO1 and IO2 (mine is the single socket wired version) via a couple of diodes and pullup resistor (to 5v).
It's just really weird that it WAS working fine as i was listening to some 3 sid and other sid tracks off the KFF after installing the arm2sid earlier....then it crapped itself when i went to go back to the menu to load something else and has been stuck since.

But one thing i didn't try for some reason...but just thought to do now, disconnected the arm2sid cable (chip still in socket) and KFF booted once or twice, but totally random and usually just garbage half readable text on screen. Also managed to run the diagnostic and it says my Phi2 freq is 985148. (PAL system)

I'm just putting it down to "something on this particular cart has gone bad...and i don't know what" as everything else is working perfectly fine. I'll figured it out one day.

uffejakobsen, 2023-03-15:
KFF needs the IO1 line working in order to get into the menu
From your statement above - I'm having a hard time figuring out if the problem is solved or not ?

Shadowtrance, 2023-03-15:
Problem isn't solved, no.

IO1 is working as everything else works no problem. In particular Sidekick64 which also has a menu like KFF (which while I can't confirm, probably uses a similar way to load it and I'm guessing needs IO1 also as it has a lot of tricks up it's sleeve).

Arm2sid also uses IO1 and IO2 which is also working as it should.

So I'm thinking something with "my" particular KFF has gone bad somewhere and not an actual system problem.
I'm just as confused as you are tbh lol

pigravity, 2023-03-15:
Personally i would blame contacts, pins somewhere on the board, USB pins or STM pins might have shortcut.

Shadowtrance, 2023-03-17:
Yeah i can't figure out what's going on with it tbh.
Something has decided to go wrong with it and it's just stuck. No amount on flashing or running over joints with the iron/flux to freshen it up makes any difference.
I've had it for about a year and a half and no real issues to speak of until now which has just left me scratching my head.

KimJorgensen, 2023-03-17:
To troubleshoot this I would try to figure out if it is the KFF, the C64 or the combination of these that is the problem.

Some things you could try if possible:
1. Try the KFF with another C64
2. Try another KFF with your C64
3. Try a cartridge that requires I/O1 e.g. a freezer or fast load cartridge with your C64
4. Reverse everything you did when installing the arm2sid and see if the KFF is working again
5. Try adjusting the phi2 clock offset as Uffe suggested
6. Measure the I/O1 signal on the expansion port using an oscilloscope or logic analyzer

Shadowtrance, 2023-03-17:
1, 2 i couldn't do (only 1 kff and 64)
Yeah i noticed earlier that freezers weren't starting with my sidekick64 but i didn't think much of it. Now it kinda makes sense.

Reversed the arm2sid install entirely (which i'd already mostly tried before but left the 2 cpu lines but sid removed and disconnected) but the diodes and resistor for the 2 IO lines were still in place.

Once fully removed i tested again, still weird the first start but got into the kff diagnostic and did nothing and it's fine again and freezers are booting on sidekick64.

So...then i started the arm2sid install again but just the 2 cpu lines (A8 / A5), R/W Pin 5 cart port.
then for the IO lines instead of doing the "both" option i just did IO2 first alone (pin 10) - KFF normal
then IO1 alone (pin 7) - communicating with usb...
and obviously if i do both IO lines with the diodes/resistor it bugs out too.

How to fix that and still be able to use IO1? I don't know but i had to figure it out and make note of it as someone else may run into it one day and pull their hair out like i have. haha

uffejakobsen, 2023-03-20:
What board revision is your "sixtyclone" based on ? 250407/250466/250469 ?
And what PLA do you use in your setup ? original 906114-01 PLA (Signetics 82S100) or a PLA replacement ?

Shadowtrance, 2023-03-20:
250466 and PLAnkton pla.
The only original chips in there now are the cia's, cpu and basic/char roms. everything else is new stock (common logic etc) or modern replacements where possible. I do have an original pla around here somewhere which I'm pretty sure works (would have to test it)

uffejakobsen, 2023-03-20:
> I do have an original pla around here somewhere which I'm pretty sure works (would have to test it)

That would be interesting to test - I know the PLAnkton is aiming for compatibility - but still the original PLA may (all together with the resistor and diodes) drive the IO1/IO2 differently...

Shadowtrance, 2023-03-21:
Same result with the original pla with full wiring for arm2sid. This is really weird. but kinda expected when we're making the system do things it probably shouldn't be doing. Something is gonna clash somewhere at some point i guess.

uffejakobsen, 2023-03-21:
If I were you I would focus on the fact that even the "disconnected" setup does not work.
That means that the just pull-up resistor and diodes alone interferes with KFF operation.
You could try with a higher resister pull-up value that the recommended 3,3k - maybe 5k-10k ?

Once that is resolved - you could look at the full solution...

As a general rule of thumb - you have to remember

You cannot have ARM2SID (also) reacting on IO1 ($DE00-$DEFF) with KFF in the system

ARM2SID addresses seems to be configurable in `address mapping configuration` screen of the ARM2SID config utility..

Also if you are to use e.g. EasyFlash emulation in KFF - then IO2 ($DF00-$DFFF) will also be off limits for ARM2SID

Generally speaking - lots of freezer carts - WarpSpeed, Final Cartridge etc etc - most of them use IO1/IO2

Having ARM2SID reacting on the same IO signals would be a problem - unless ARM2SID access pattern to IO1/IO2 is extremely clever.

pazuzu72, 2023-07-10:
I don't know if it can be useful. 
I get the "Communicating with usb" message with firmware 1.44 when I press **SHIFT + RETURN** and choose **LOAD** to run a **PRG**. If I use firmware **1.43** all ok instead. All chips are original and working, assy **250425** (tested on 4 C64), on the **C64C** instead no problem with **1.44**.

closed 2024-12-29 by KimJorgensen