KimJorgensen/KungFuFlash #165: Small changes for ef3utils, ef3bbs, and libef3usb to compile on Apple Silicon

fortiko, 2023-04-27:
Hello! 

With these simple modifications, I can compile ef3utils, ef3bbs, and libef3usb on macOS Apple Silicon. 

It would be great if you could please incorporate those changes. Thank you for your consideration!

```
% git diff
diff --git a/3rd_party/ef3utils/Src/ef3usb.c b/3rd_party/ef3utils/Src/ef3usb.c
index 55a78c1..5e00978 100644
--- a/3rd_party/ef3utils/Src/ef3usb.c
+++ b/3rd_party/ef3utils/Src/ef3usb.c
@@ -660,7 +660,7 @@ int main(int argc, char *argv[])
   buf = (unsigned char *) malloc(0x1000000);
   filebuf = (unsigned char *) malloc(0x1000000);
   bufread = (unsigned char *) malloc(30);
-  bufstart = (unsigned char *) malloc(30);
+  bufstart = ( char *) malloc(30);

   if (fname != NULL && (command != 3 && command != 11))
   {
diff --git a/3rd_party/ef3utils/Src/taper.tas b/3rd_party/ef3utils/Src/taper.tas
index db8f175..f90f381 100644
--- a/3rd_party/ef3utils/Src/taper.tas
+++ b/3rd_party/ef3utils/Src/taper.tas
@@ -39,7 +39,7 @@ end
   rts
 .pend
 ;            0123456789012345678901234567890123456789
-.enc screen
+.enc "screen"
 text1  .text "initiating tap transfer now...",0

 ;---------------------------------
@@ -219,7 +219,7 @@ end4
   jmp endtaptransfer
 .pend
 ;            0123456789012345678901234567890123456789
-.enc screen
+.enc "screen"
 text2  .text "* end of tap file.",0
 text3  .text "* an error occured during transfer!",0
```

KimJorgensen, 2023-07-27:
`taper.tas` has now been updated for the latest 64tass compiler version, thank you.

However, I'm not sure why the cast in `ef3usb.c` should be changed? `bufstart` is declared as an `unsigned char *`

fortiko, 2023-07-29:
@KimJorgensen you can certainly leave out the cast, I was just adding it to avoid several warning of this type:

`passing 'unsigned char *' to parameter of type 'const char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]`:

```
cc -O2 -o ../ef3usb ef3usb.c
ef3usb.c:740:34: warning: passing 'unsigned char *' to parameter of type 'const char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
                if (!(fp = fopen(bufstart, "wb")))
                                 ^~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdio.h:153:37: note: passing argument to parameter '__filename' here
FILE    *fopen(const char * __restrict __filename, const char * __restrict __mode) __DARWIN_ALIAS_STARTING(__MAC_10_6, __IPHONE_2_0, __DARWIN_ALIAS(fopen));
                                       ^
ef3usb.c:861:41: warning: passing 'unsigned char *' to parameter of type 'char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
                                sprintf(bufstart,"%2d %2d",track,sector);
                                        ^~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/secure/_stdio.h:47:28: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                           ^~~
ef3usb.c:1059:41: warning: passing 'unsigned char *' to parameter of type 'char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
                                sprintf(bufstart,"%2d %2d",track,sector);
                                        ^~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/secure/_stdio.h:47:28: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                           ^~~
ef3usb.c:1356:25: warning: passing 'uint8_t[256]' (aka 'unsigned char[256]') to parameter of type 'char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
                sprintf(buffer, "C64-TAPE-RAW%c", 0x01);
                        ^~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/secure/_stdio.h:47:28: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                           ^~~
ef3usb.c:1499:25: warning: passing 'unsigned char *' to parameter of type 'char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
                sprintf(bufstart,"EFSTART:PRG%c",0);
                        ^~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/secure/_stdio.h:47:28: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                           ^~~
ef3usb.c:1576:25: warning: passing 'unsigned char *' to parameter of type 'char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
                sprintf(bufstart,"EFSTART:CRT%c",0);
                        ^~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/secure/_stdio.h:47:28: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                           ^~~
6 warnings generated.

fortiko, 2023-07-29:
With your original declaration of `bufstart`  ef3usb compiles and runs on macOS 13 Ventura and Xcode 14

KimJorgensen, 2023-08-07:
The `bufstart` warnings should be fixed now and I'll close the issue, but let me know if I missed something

closed 2023-08-07 by KimJorgensen