KimJorgensen/KungFuFlash #166: IDEA: Kung Fu Flash SID emulator

Sgw32, 2023-05-03:
Hello,

Before I've made an UltiSID emulator, which is, well, good on simple tunes and runs on STM32F103, but now I've made a new firmware for it for STM32F405RGT6 based on Kung Fu Flash firmware. 
Take a look, maybe someone is interested: 
https://github.com/Sgw32/KungFuSID

Now your Kung Fu Flash cartridge even can emulate SID inside C64 😄 
Emulation is cycle-exact on 0xD400, but the firmware is still in beta stage, needs some optimization. My goal is to make a new UltiSID on Kung Fu SID firmware to make an open source ArmSID alternative.

mamejay, 2023-05-05:
This is exciting.  Will the UltiSID have support for paddles?

Sgw32, 2023-05-15:
I'll try to do it. Probably it just needs ADC connection to POTXY pins.

KimJorgensen, 2024-12-29:
Thanks for sharing

closed 2024-12-29 by KimJorgensen