KimJorgensen/KungFuFlash #167: Did development stop?

Gabeki17, 2023-05-08:
Hi,

Are there future plans might be?

Thx,
Gábor

pigravity, 2023-05-11:
hope not.

DWStaley, 2023-06-27:
If they're still supporting this project, it's a rather slow process. I submitted a new issue 6 weeks ago and still have no initial response from anyone.

pigravity, 2023-06-27:
Would be very unfortunate and sad, there is alot of people now using KungFu and loving it, unless some other cart will take its place like SideKick or something, but still... I prefer KungFu :)

ogg1e, 2023-06-27:
These people do this stuff on their own free time.  We're lucky this exists at all.

KimJorgensen, 2023-07-26:
Development and answering support questions is done for free in my limited spare time. Feel free to fork this project and add features if you get impatient

closed 2023-07-26 by KimJorgensen