KimJorgensen/KungFuFlash #168: Block Write does not write the buffer pointer

DWStaley, 2023-05-15:
I recently observed saving the high scores sector (T12/S7) after completion of a Lode Runner session, the high scores no longer loaded into the game (i.e., high scores display is missing all entries). Using SMON I discovered the ExecDiskCmdWrite in Load Runner assumed the disk drive would automatically skip the first byte of the sector (which is the buffer pointer in a direct access B-W command) and start with the second byte. The KFF does not skip this byte, however. (The SD2IEC does.) Thus when the high scores and the disk ID are written to the disk sector, they are offset by 1 byte, making it impossible for them to load back in. When I added code to the Lode Runner game to write the buffer pointer byte before sending the rest of the data, the problem was resolved.

This information is also noted in the 1541-II disk drive user's guide.

FYI: Lode Runner uses the U1/U2 block read/write commands.

I noticed the direct access block read/write routines were added to KFF last year. Perhaps this got overlooked.

KimJorgensen, 2023-08-11:
Yes, it looks like the written block is offset by one byte.
However, as you wrote, Lode Runner does not use the "B-R" or "B-W" command (and these are currently not supported by KFF) so it must be some other incompatibility that causes this offset

KimJorgensen, 2023-09-23:
After some debugging it turned out that opening a direct access buffer on a 1541 will set the buffer pointer to 1 (skipping the first byte) whereas KFF would set it to 0.
This is a problem when writing to the channel without using the buffer pointer (B-P) or block read (U1) command first.

I'll get that fixed in the next release

KimJorgensen, 2023-09-23:
Should be fixed in the latest [firmware release](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.48)

closed 2023-09-23 by KimJorgensen