KimJorgensen/KungFuFlash #173: Save the file on a cassette tape

RetroChlop, 2023-08-08:
Would it be possible to make such a software for KFF so that we can enter a given game, select the PRG file of interest and save it to a cassette tape ? 
Of course, it would be welcome if there would be an option in the sofet to save the file in normal or turbo. So far, I create compilations on cassettes using TURBO cartridge and PI 1541 EPYX FAST LOAD in which I load the game into the computer memory, then use LIST command to open its contents and use ⬅️"GAME NAME" Return command to save the file on the cassette.

KimJorgensen, 2023-08-11:
Yes, that would be a nice enhancement.

With the current firmware, you can load software from the SD card in BASIC (using the built-in drive emulator) and save that to tape in normal mode. It is also possible to transfer .TAP files via USB and ef3utils.

RetroChlop, 2023-08-11:
The issue of transferring files via USB has already been tested by me. It
would be nice to add such an option to KFF that would allow you to enter a
file, e.g. D64 or T64, and if the game is single-file then record it to a
cassette in turbo mode. With PRG files it is much easier because you don't
have to open it but can record it directly. Overall, I am very satisfied
with the use of KFF and its available features 😁.

RetroChlop, 2023-11-30:
> I also agree this would be a fun feature to have in KFF.

You can think about it 😎 Probably not one KFF user would be happy 😎