KimJorgensen/KungFuFlash #174: Possible issue with the new J CIA replacement

lordoftears, 2023-08-22:
I have tried the new J Cia replacements on my C64 and I am experiencing some issues with my Kung Fu Flash cartridge: it does not load the prgs and the menu screen is a mess of random characters. I have tried both with the new (8521) and old (6526) CIA configurations.
Anyone else is experiencing this issue with J-CIA and KFF?

KimJorgensen, 2023-08-23:
I have not tried the J-CIA replacement but have you tried to adjust the [phi2 offset](https://github.com/KimJorgensen/KungFuFlash/blob/master/firmware/README.md#diagnostic)? It could be that the timing requirements are slightly different compared to the original MOS CIA.

Otherwise this suggest that the replacement J-CIA is not 100% compatible with the original (assuming that Kung Fu Flash works fine with the latter) and that this issue should be reported to the maintainer of J-CIA

lordoftears, 2023-08-23:
With a +2 offset the issues seem to have gone. Thank you

closed 2023-08-23 by KimJorgensen