KimJorgensen/KungFuFlash #176: No Effect trying to update.

rafael1138, 2023-09-10:
Hi.

I have a KungFu Flash from Retrorewind with V1.44. I have downloaded the V1.47 and the system shows the upgrade menu and I selected Upgrade. After that I get a partial screen with the do not turn off message but inm half second returns to the main screen and the Versionat the botton still shows V1.44. I have tried V1.45 and V1.46 and the same issue.

rafael1138, 2023-09-10:
Just want to add that the KFF is working with v1.44. Is just that is not allowing any update.

KimJorgensen, 2023-09-11:
@rafael1138 I'm not sure why you can't upgrade the firmware but I think you should contact the seller with this issue.

I have no problem with people selling Kung Fu Flash cartridges, but they should also be willing to handle the support if something doesn't work

closed 2023-09-11 by rafael1138