KimJorgensen/KungFuFlash #178: Native Freezer/Restore feature

mwedmark, 2023-09-27:
Maybe someone already asked. KFF can only do one thing at a time: either being Freezer Cartridge, being a D64-reader/writer or being a SD-loader.  I guess this is both up to 64KB RAM and maybe partly PCU speed.

I got a question about using KFF as a freezer and tried it out. It works. 
For example using:
- KFF as Final Cartridge III, exiting to BASIC with FC3 enabled
- Reading/Starting a game from my PI-1541. 
- Then mounting an empty disk to PI-1541
- Running the game up to a point where you want to save.
- Pressing Freeze and choosing FDISK command from Freezer menu. _FC and FC is saved into disk
- Then later loading and starting the file FC will bring back a game to the point it was saved/freezed.

Is it possible to implement a feature so that the Freezer button by default freezes/dumps memory to FC-like file/s to SD-card, maybe selecting name of file and then returning to game? That would be a cool and useful features for those games cheaters out there.

KimJorgensen, 2023-09-28:
Yes, it would be a really cool feature to have a built in freeze and restore functionality and even a monitor for hacking.

Let me know if anyone knows of a good and reliable open source freezer that is compatible with the KFF license

magwed, 2023-09-28:
Backbit has native support for simple save/restore. I'm not sure it is open but I guess it should be possible to check out different solutions, learn and write our own using information learned. Also I would guess parts of this is in VICE which maybe have a resonable license.

mwedmark, 2023-10-10:
Here are my suggestions for this feature from a crude "what I want" perspective, not what is possible:
- Default Kung Fu Freezer/Dumper can be activated by pressing Freezer button when no other specific Freezer cartridge is active. Could possibly be disabled in settings? (Disable default Kung Fu Freezer)
- Should put C64/CPU in stable halt mode (as described in papers availble)
- On halt, freeze code should be executed from external Cartridge mem (FLASH?) and back up full mem to file(f.e. *.KDP) on SD-card in current directory(the place where the last thing was started/loaded from). Saving 64KB RAM/Stack, All registers inc. PC. At end return to KFF menu and show the files that just were created.
- Potentially KFF can keep track of if this file has been dumped before and just increase name of file using enumeration: Last app started and added enum at end.. or just save over the old file similar to what is happening when save EasyFlash cart that has been changed.
- Possible actions on KDP-file in menu system:
    - COPY
    - RENAME
    - DELETE
    - RESTORE
- Do we need / want a prg that could start the KDP files? Depends on if you want it to be restored without having a KFF Cartridge. Possible?
- Maybe add a simple monitor as well, running from Flash mem. Short press dumps mem to disk, long press starts monitor for peek/poke/coding and also give you commands for dumping before/after changing mem or just restart program from where it was halted.

Tissemon, 2024-01-10:
The ability to save-state with KFF would be truly amazing! I don't know if it's possible, but if it would be possible to save states directly to the SD-card would probably be the best and fastest approach. With OG freezer-carts you have to hazzle with changing disks to an empty disk before saving, and also use fast-loading to load it back to the memory as fast as possible.
Maybe, if it's possible, dump everything from memory to a file that saves to /savestates/activegame01.sav. When you load activegame.crt the next time, it checks if there is a sav-file available and asks if you want to load. You then mount the file as usual and also restores everything to exactly the memory positions stored in the .sav-file. Or you just load the game from the start as usual and press freeze again, but this time you get the option to either save or load a savestate.
As I said, don't know if this approach is possible with the KFF, but would be nice if it was :-)

Also, if you havn't seen it already, the source code for Action Replay seems to be available here: https://rr.pokefinder.org/wiki/Action_Replay (at the bottom of the page you can find the file Action_Replay_all_rr.c64.org_2022-08.rar that contains an AR_Sources-folder)
You might be able to find out how the freezer in that cart works through that code.

beatmaster13, 2024-08-31:
Hi Kim! Besides what was described earlier, what is the possibility of natively adding a VICE-style debugging monitor?