KimJorgensen/KungFuFlash #179: Kung Fu Flash black/grey screen issue

steswart, 2023-11-01:
I have just built 4 x Kung Fu Flash cartridges using a STM32F405RGT6 chip. I managed to program the arm chip with the v1.48 firmware using DfuseDemo, but when I insert the cartridge in the c64 I get a black/grey screen. Could this be a fake STM32F405RGT6 chip I got from AliExpress? All 4 shows the same issue. The jumpers are not connected so it is not that.
Please let me know where I must start looking to get it working?

KimJorgensen, 2023-11-01:
I have no experience with fake or non STM chips, but if the LED is dead it could suggest that the chip wasn't programmed correctly.

For programming I would recommend a ST-Link V2 programmer, but you could check [these instructions](https://github.com/KimJorgensen/KungFuFlash/issues/70#issuecomment-753661669) to see if you did the same

steswart, 2023-11-01:
Hi KimJorgensen,

Thanks for the help. I followed the instructions using DFUDemo. Now the led turns on and reset button works, but i get no display. Same black/grey screen.
Could this be that am using a 22pf cap at c7 and c8 instead of a 20pf? I have ordered 20pf caps. Also getting a St-Link tomorrow

KimJorgensen, 2023-11-01:
I'm not sure about C7 and C8, but if the LED turns on it indicates that the clock for the MCU is running.

Did you test with all 4 cartridges after reprogramming them? Else I would check the reset line to the C64 to see if it held low by the MCU.
If you have access to another C64 you could also test with that

steswart, 2023-11-02:
Hi Kim,

I reloaded the firmware using a ST-Link v2 on all 4, found some loose ARM legs on one and a badly soldered resistor array leg on another. So, I got 3 of the 4 working now. Will look at the 4th one later.
Thanks for your help 😊

closed 2023-11-02 by steswart